jneilliii/OctoPrint-WemoSwitch #1: First Attempts at Testing:

talldonkey, 2018-09-08:
Okay, so my first attempts to run run was not successful. I have two printers hooked up to two different Wemo Insights and configured the IP Addresses, in two separate raspberry Pi's running Octoprint. The symptoms of both unsuccessful runs have the lightning bolt just spinning but nothing happening. I did turn on debug logging for both.

Looks like both instances are not able to connect.  These IP Addresses were sourced from the official wemo android app's settings/hardware display.

`[2018-09-08 19:48:18,602] DEBUG: Checking status of 192.168.0.117.
[2018-09-08 19:48:18,603] DEBUG: IP 192.168.0.117 is valid.
[2018-09-08 19:48:18,604] DEBUG: Attempting to connect to 192.168.0.117
[2018-09-08 19:48:45,869] DEBUG: Checking status of 192.168.0.117.
[2018-09-08 19:48:45,870] DEBUG: IP 192.168.0.117 is valid.
[2018-09-08 19:48:45,870] DEBUG: Attempting to connect to 192.168.0.117
[2018-09-08 19:49:12,083] DEBUG: Checking status of 192.168.0.117.
[2018-09-08 19:49:12,084] DEBUG: IP 192.168.0.117 is valid.
[2018-09-08 19:49:12,085] DEBUG: Attempting to connect to 192.168.0.117
[2018-09-08 19:54:17,452] DEBUG: Checking status of 192.168.0.117.
[2018-09-08 19:54:17,453] DEBUG: IP 192.168.0.117 is valid.
[2018-09-08 19:54:17,454] DEBUG: Attempting to connect to 192.168.0.117
`
`[2018-09-08 19:52:41,801] DEBUG: Checking status of 192.168.0.134.
[2018-09-08 19:52:41,802] DEBUG: IP 192.168.0.134 is valid.
[2018-09-08 19:52:41,803] DEBUG: Attempting to connect to 192.168.0.134
`

jneilliii, 2018-09-08:
Ok, so it looks like it sees the device via IP. At least that part works. When developing I was using wemo emulation mode on a tasmota device, so I did fear this would happen. I may have to purchase a real device though to move forward with development... or do a bunch of searching. Are there any errors in octoprint.log?

jneilliii, 2018-09-08:
If you open a browser on the same network and go to the addresses listed below, does it return information or prompt for login, etc.?

http://192.168.0.117/setup.xml

https://192.168.0.117/setup.xml

Just trying to figure out if it's a discovery issue or a command sending issue.

talldonkey, 2018-09-10:
Checked Octoprint.log, and found this:

I also tried the WeMo Mini that I have at IP address .148 you see, same apparent symptoms, the capture below (octoprint.log) is from the WeMo Mini Attempt.  The wemo plugin log looks exactly same as already posted (but with .148 IP address)

~~~
2018-09-10 09:36:09,520 - octoprint - ERROR - Exception on /api/plugin/wemoswitch [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1140, in decorated_view
    return flask_login.login_required(func)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask_login.py", line 758, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 99, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_wemoswitch/__init__.py", line 143, in on_api_command
    self.check_status("{ip}".format(**data))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_wemoswitch/__init__.py", line 121, in check_status
    chk = self.sendCommand("info",plugip)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_wemoswitch/__init__.py", line 170, in sendCommand
    device = pywemo.discovery.device_from_description("http://%s/setup.xml" % plugip, None)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pywemo/discovery.py", line 40, in device_from_description
    xml = requests.get(description_url, timeout=10)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/api.py", line 72, in get
    return request('get', url, params=params, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/api.py", line 58, in request
    return session.request(method=method, url=url, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/sessions.py", line 508, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/raven/breadcrumbs.py", line 326, in send
    resp = real_send(self, request, *args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/sessions.py", line 618, in send
    r = adapter.send(request, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/adapters.py", line 508, in send
    raise ConnectionError(e, request=request)
ConnectionError: HTTPConnectionPool(host='192.168.0.148', port=80): Max retries exceeded with url: /setup.xml (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x6b36b550>: Failed to establish a new connection: [Errno 111] Connection refused',))
2018-09-10 09:36:09,524 - tornado.access - ERROR - 500 POST /api/plugin/wemoswitch (::ffff:192.168.0.111) 76.05ms
2018-09-10 09:36:20,205 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.0.111) 26.10ms
~~~

talldonkey, 2018-09-10:
As for the 
http://192.168.0.148/setup.xml
https://192.168.0.148/setup.xml

I get a:

"This site can’t be reached
192.168.0.148 refused to connect.
Search Google for set up
ERR_CONNECTION_REFUSED"

jneilliii, 2018-09-10:
Thanks. It's a device discovery issue then it appears with the pywemo commands. May have to change how that finds the plug for real world example. Are you comfortable connecting ssh to your pi and running linux commands if I were to give them to you?

talldonkey, 2018-09-10:
Yes, comfortable running commands via ssh to the pi.

jneilliii, 2018-09-10:
Please try the following commands.

```
cd ~/oprint/bin/
./python
import pywemo
devices = pywemo.discover_devices()
print(devices)
```

That will tell us if the standard pywemo discovery process will work since the direct ip method seems to be blocked. If it works then there should be two devices returned in the print command. Run the following and it should toggle the first one in the list.

```
devices[0].toggle()
```

Once done you can close out of python holding down the CTRL key and pressing the letter Z on your keyboard.

beneaths, 2018-09-12:
I can confirm that doing the above will toggle the Wemo device on and off. 

When I listed devices - it only listed the Wemo device with the ID that I specified in the wemo app. If you need any more troubleshooting, I'll help where I can.

talldonkey, 2018-09-12:
The discovery revealed all three wemo devices I have set up in the official app.  2 Insights, and 1 Wemo Mini switch.

Toggling all three of them were successful - was able to turn on and off all three via the toggle() command.

jneilliii, 2018-09-12:
Do you guys know if it's possible to have multiple devices in the app with the same name or does it force you to make them distinct?  That's the only possible issue I can think of with this revised approach.

talldonkey, 2018-09-12:
Yes it's possible to have multiple devices with the same name in App.
 Just tested with the mini and insight and changed the name to same as
another and the app allowed it.

jneilliii, 2018-09-14:
Ok, thanks @talldonkey.  That's going to complicate things a bit, but as long as I can instruct people that duplicate names could be an issue I don't think it will be that much of a problem.  My thought process is to loop through the list of devices found during the discovery command and the first one that matches the name associated with the button in the plugin settings will be the one that it will control.  My internet went down last night and has yet to come back up (dump truck took out the pole down the street) but I plan on working on this plugin some this weekend.

jneilliii, 2018-09-15:
Ok guys, I've just pushed an update that uses the device discovery rather than a direct ip connection.  On my emulated device it is working as expected.  You'll need to adjust your settings replacing the ip address with the device name you have configured in the app. Make sure OctoPrint restarts and the web interface reloads, otherwise force the browser reload by pressing ctrl+f5 to refresh the local cache.

[image]

If you have issues enable debug logging in settings and post up the plugin_wemoswitch_debug.log file.

talldonkey, 2018-09-15:
Excellent - Works perfectly on both of my printers connected to each of their own Wemo Insight switches!

bolsoncerrado, 2018-09-27:
on the discover_devices() i got

"No handlers could be found for logger "pywemo.ssdp""

but the print(devices) indeed printed their exact names.

totic, 2018-09-29:
Hi I am also getting the problem "**Wemo Switch Error**" whenever I click on the icon.

When I run the python code you posted above I see:

`>>> print(devices)
[<WeMo Switch "camera">, <WeMo Insight "printer">]`

I configured the printer the following way:

[image]

should the printer name in the settings be exactly `<WeMo Insight "printer">`, because I tried that one and also `WeMo Insight "printer"`, `"printer"` and `printer` none of them work

jneilliii, 2018-09-29:
Can you enable debug logging and restart octoprint and then upload your `plugin_wemoswitch_debug.log` file please?

Also if you run these commands in ssh what is returned? Make sure to replace the ip with the ip of your insight printer.

```
cd ~/oprint/bin/
./python
import pywemo
address = "192.168.1.50"
port = pywemo.ouimeaux_device.probe_wemo(address)
url = 'http://%s:%s/setup.xml' % (address, port)
url = url.replace(':None','')
print(url)
device = pywemo.discovery.device_from_description(url, None)
print(device)
```

jneilliii, 2018-09-29:
@totic, I just pushed a pre-release version to a separate branch that may resolve your issue.  Install using the following url in plugin manager and report back.  Once installed you'll need to update your settings to use the IP address of the plug instead of device name.

https://github.com/jneilliii/OctoPrint-WemoSwitch/archive/0.1.1.zip

totic, 2018-09-29:
@jneilliii that worked, but with a bug, it turns it on, but the UI said there was an error 
(but the wemo device did turn on) when I press it again it turns green and then the third time it does turn red, so when I click on it I get: 
1st Click: Error (but does turn on), 
2nd Click: Turns green but doesnt do anything
3rd Click: Turns it off

Thanks again!

totic, 2018-09-29:
It only happens when its off that I have to click it twice, for the UI to match the state of the printer. 
Here is the UI output: 

[image]

And the logs:
[plugin_wemoswitch_debug.log](https://github.com/jneilliii/OctoPrint-WemoSwitch/files/2430863/plugin_wemoswitch_debug.log)

talldonkey, 2018-09-29:
I've also noticed that sometimes the UI doesn't match the state of the
switch, despite having polling on.  Clicking on it a few times does update
it.

For example, if Wemo switch is actually on, and UI shows red lightning
bolt, indicating it is off.
Clicking lightning bolt once will update it to green.  Wemo Switch stays
on, state unchanged.
Clicking lighting bolt again will turn off Wemo switch.

Same in reverse.  Click once to refresh, then it'll show state.  The
theoretical risk is, if the state is correct in the first place, one may
accidentally turn on/off the printer when not intended, but there are so
many ways to see if printer is on in octoprint this in practice I've found
this not a concern if one is "awake and paying attention"

Haven't gotten around to reporting it as it was only a very minor thing
with easy workarounds.  Note that I'm using the release version using Wemo
Device names not IP names.

jneilliii, 2018-09-29:
Thanks @totic, that is showing me that the plug is returning a status of 8, which I'm not accounting for.  I need to do some research to see what that is supposed to mean.  I expect either 0 (off) or 1 (on).  Can you guys run the following commands on your pi (replacing the ip) for me again please in between toggling it on and off from the OctoPrint side?

```
cd ~/oprint/bin/
./python
import pywemo
address = "192.168.0.104"
port = pywemo.ouimeaux_device.probe_wemo(address)
url = 'http://%s:%s/setup.xml' % (address, port)
url = url.replace(':None','')
print(url)
device = pywemo.discovery.device_from_description(url, None)
print(device)
print(device.get_state())
```

jneilliii, 2018-09-29:
I think I found out that state.  The insight devices include a state of 8 that means it's on but load is off, https://ouimeaux.readthedocs.io/en/latest/_modules/ouimeaux/device/insight.html?highlight=state.  Let me make a quick adjustment to the code and re-upload.

jneilliii, 2018-09-29:
Ok, just pushed an update that is still working with my emulated device.  It should in theory resolve this issue with the insight device.  Give it some testing and let me know.  Install from the same url, below for convenience. Make sure OctoPrint restarts after install and manually force refresh the browser side by pressing CTRL+F5.

https://github.com/jneilliii/OctoPrint-WemoSwitch/archive/0.1.1.zip

totic, 2018-09-30:
That seems to have fixed it !!!! thanks so much

talldonkey, 2018-09-30:
Interesting, my previously working setup with 0.1.0 with Wemo Names are no longer working on both of the printer. I am also getting the problem "Wemo Switch Error" whenever I click on the icon.  Previous to that I would not get error messages.  I'm a bit surprised because I didn't touch the working configuration.

`[2018-09-30 22:43:18,697] DEBUG: Checking status of Ender.
[2018-09-30 22:43:18,699] DEBUG: Attempting to find Ender
[2018-09-30 22:43:18,700] DEBUG: [<WeMo Switch "Filament Dryer">, <WeMo Insight "Ender">, <WeMo Insight "Prusa">]
[2018-09-30 22:43:18,701] DEBUG: Filament Dryer found.
[2018-09-30 22:43:18,701] DEBUG: Ender found.
[2018-09-30 22:43:18,702] DEBUG: Sending command info to Ender
[2018-09-30 22:43:18,702] DEBUG: 8`

I'll try version 0.1.1 and put in IP addresses and see if that works better but I'm confused as to what 'broke'- Did Wemo make a change?

jneilliii, 2018-09-30:
That was another fix I merged into the latest version. Insight devices have a return status of 8 that I was not accounting for, which from what I understand means the plug is on but the is no load.

talldonkey, 2018-10-10:
Update - Been working perfectly with IP addresses. :)