jneilliii/OctoPrint-WemoSwitch #2: Feature Request : AutoShutdown after delay

noxin1, 2018-09-18:
A timed delay where the outlet/printer is shut off.

jneilliii, 2018-09-18:
I think this will work; if you enable auto disconnect and specify a delay, the power off will wait that many seconds before sending the command to the plug.  So the following would power off the plug 3 minutes after receiving the shutdown command.

[image]

noxin1, 2018-09-18:
Oh!  I thought that was just going to disconnect from the printer, not shut it down.   I scanned the code but didnt see it send a shutoff command, probably missed it!     Thanks for the quick response!    This means I can stop using my hacky PSU Control setup

jneilliii, 2018-09-19:
In theory it should work that way.  There is a sleep command that basically pauses python execution if my understanding is correct and then continues through the code after that configured delay.  The line in question is [here](https://github.com/jneilliii/OctoPrint-WemoSwitch/blob/24d3220b03027c828b71c41a5613c2d50f369710/octoprint_wemoswitch/__init__.py#L115).  Let me know if doesn't work as expected.  It's not an idle timeout though, you still have to send the power off command, which can be automated with the gcode scripts at the end of the print.

closed 2018-09-19 by jneilliii