jneilliii/OctoPrint-WemoSwitch #3: Feature Request - Thermal shutdown

dgarlans, 2018-09-18:
Is there a way to connect the plugin to Octoprint's thermal panel, so that if the bed temperature goes into runaway somehow, the plugin can trigger a wemo to kill the power to the printer?

My printer failed hot one time and I walked into the room with an 120c bed, and it would be awesome to have a way to prevent that.

jneilliii, 2018-09-18:
Sorry, but not familiar with the thermal panel you are referring to? As it stands now, no there is no events outside of GCODE commands to trigger an on/off of the wemo device.

colemorgan, 2018-10-01:
@jneilliii could you monitor the printer temp responses and create a threshold that way?

jneilliii, 2018-10-01:
Yeah, that would be an option.  I'll see about merging that in on the next updates I do on this plugin.

Webstas, 2018-10-23:
@jneilliii could you merge that into your TPLink plugin as well?

jneilliii, 2019-04-20:
New release version ready that incorporates thermal runaway protection.  Please test and let me know hoe it goes.  Install using the following url in plugin manager.

https://github.com/jneilliii/OctoPrint-WemoSwitch/archive/0.1.3.zip

closed 2019-05-05 by jneilliii

jediknight2, 2019-06-10:
I just tried to test it and it doesn't seem to be working.  I'm willing to work with you to get it working as I have a 120V bed and the SSR failed the other day. It would be AWESOME for OctoPrint and your Plugin to see the temperature going high and trigger the OFF on the WeMo switch.

jediknight2, 2019-06-10:
Here is what I am seeing in log.  I can turn power on and off via the WeMo app

[2019-06-10 19:35:14,816] DEBUG: Checking status of 192.168.0.15.
[2019-06-10 19:35:14,817] DEBUG: IP 192.168.0.15 is valid.
[2019-06-10 19:35:14,817] DEBUG: Attempting to connect to 192.168.0.15
[2019-06-10 19:35:24,837] DEBUG: Getting device info from http://192.168.0.15/setup.xml
[2019-06-10 19:35:41,882] DEBUG: Checking status of 192.168.0.15.
[2019-06-10 19:35:41,883] DEBUG: IP 192.168.0.15 is valid.
[2019-06-10 19:35:41,883] DEBUG: Attempting to connect to 192.168.0.15
[2019-06-10 19:35:51,902] DEBUG: Getting device info from http://192.168.0.15/setup.xml
[2019-06-10 19:36:33,770] DEBUG: Max bed temp reached, shutting off plugs.
[2019-06-10 19:36:33,774] DEBUG: Turning off 192.168.0.15.
[2019-06-10 19:36:33,777] DEBUG: {'autoConnectDelay': 10, 'autoConnect': True, 'sysCmdOn': False, 'displayWarning': True, 'ip': '192.168.0.15', 'btnColor': '#808080', 'sysRunCmdOn': '', 'sysRunCmdOff': '', 'autoDisconnect': False, 'label': '3D Printer', 'autoDisconnectDelay': 0, 'sysCmdOff': False, 'thermal_runaway': True, 'sysCmdOffDelay': 0, 'gcodeOffDelay': 1, 'sysCmdOnDelay': 0, 'gcodeEnabled': False, 'gcodeOnDelay': 1, 'currentState': 'unknown', 'warnPrinting': True, 'icon': 'icon-bolt'}
[2019-06-10 19:36:33,778] DEBUG: IP 192.168.0.15 is valid.
[2019-06-10 19:36:33,787] DEBUG: Attempting to connect to 192.168.0.15
[2019-06-10 19:36:35,777] DEBUG: Max bed temp reached, shutting off plugs.
[2019-06-10 19:36:35,780] DEBUG: Turning off 192.168.0.15.
[2019-06-10 19:36:35,783] DEBUG: {'autoConnectDelay': 10, 'autoConnect': True, 'sysCmdOn': False, 'displayWarning': True, 'ip': '192.168.0.15', 'btnColor': '#808080', 'sysRunCmdOn': '', 'sysRunCmdOff': '', 'autoDisconnect': False, 'label': '3D Printer', 'autoDisconnectDelay': 0, 'sysCmdOff': False, 'thermal_runaway': True, 'sysCmdOffDelay': 0, 'gcodeOffDelay': 1, 'sysCmdOnDelay': 0, 'gcodeEnabled': False, 'gcodeOnDelay': 1, 'currentState': 'unknown', 'warnPrinting': True, 'icon': 'icon-bolt'}
[2019-06-10 19:36:35,785] DEBUG: IP 192.168.0.15 is valid.
[2019-06-10 19:36:35,786] DEBUG: Attempting to connect to 192.168.0.15
[2019-06-10 19:36:37,774] DEBUG: Max bed temp reached, shutting off plugs.
[2019-06-10 19:36:37,779] DEBUG: Turning off 192.168.0.15.
[2019-06-10 19:36:37,782] DEBUG: {'autoConnectDelay': 10, 'autoConnect': True, 'sysCmdOn': False, 'displayWarning': True, 'ip': '192.168.0.15', 'btnColor': '#808080', 'sysRunCmdOn': '', 'sysRunCmdOff': '', 'autoDisconnect': False, 'label': '3D Printer', 'autoDisconnectDelay': 0, 'sysCmdOff': False, 'thermal_runaway': True, 'sysCmdOffDelay': 0, 'gcodeOffDelay': 1, 'sysCmdOnDelay': 0, 'gcodeEnabled': False, 'gcodeOnDelay': 1, 'currentState': 'unknown', 'warnPrinting': True, 'icon': 'icon-bolt'}
[2019-06-10 19:36:37,784] DEBUG: IP 192.168.0.15 is valid.
[2019-06-10 19:36:37,785] DEBUG: Attempting to connect to 192.168.0.15
[2019-06-10 19:36:39,780] DEBUG: Max bed temp reached, shutting off plugs.
[2019-06-10 19:36:39,783] DEBUG: Turning off 192.168.0.15.
[2019-06-10 19:36:39,786] DEBUG: {'autoConnectDelay': 10, 'autoConnect': True, 'sysCmdOn': False, 'displayWarning': True, 'ip': '192.168.0.15', 'btnColor': '#808080', 'sysRunCmdOn': '', 'sysRunCmdOff': '', 'autoDisconnect': False, 'label': '3D Printer', 'autoDisconnectDelay': 0, 'sysCmdOff': False, 'thermal_runaway': True, 'sysCmdOffDelay': 0, 'gcodeOffDelay': 1, 'sysCmdOnDelay': 0, 'gcodeEnabled': False, 'gcodeOnDelay': 1, 'currentState': 'unknown', 'warnPrinting': True, 'icon': 'icon-bolt'}
[2019-06-10 19:36:39,787] DEBUG: IP 192.168.0.15 is valid.
[2019-06-10 19:36:39,788] DEBUG: Attempting to connect to 192.168.0.15
[2019-06-10 19:36:41,916] DEBUG: Checking status of 192.168.0.15.
[2019-06-10 19:36:41,917] DEBUG: IP 192.168.0.15 is valid.

jneilliii, 2019-06-12:
@jediknight2 does the plug turn on and off with the button in the navbar?

jediknight2, 2019-06-13:
No, icon just spins.

[2019-06-12 22:20:49,911] DEBUG: Checking status of 192.168.0.15.
[2019-06-12 22:20:49,916] DEBUG: IP 192.168.0.15 is valid.
[2019-06-12 22:20:49,917] DEBUG: Attempting to connect to 192.168.0.15
[2019-06-12 22:20:59,939] DEBUG: Getting device info from http://192.168.0.15/setup.xml

PR #17 (2019-05-05): 0.1.3
- Added thermal runaway monitoring, resolves #3.
- Changed default gcode off delay to 1 second to avoid conflicts with warn while printing option, resolves #10.