jneilliii/OctoPrint-WemoSwitch #5: the icon just spins....

DuaneLitchfield, 2018-09-26:
Hi, I have installed the addin and configured it as instructed.

However, when I click on the icon to toggle the wemo plug state the icon just spins around.

I enabled the log and this is what is in there.
[2018-09-26 18:54:35,484] DEBUG: Checking status of Printer.
[2018-09-26 18:54:35,484] DEBUG: No devices cached, discovering devices.
[2018-09-26 18:54:38,293] DEBUG: Checking status of Printer.
[2018-09-26 18:54:38,294] DEBUG: No devices cached, discovering devices.

I have given the wemo a static i and also tried that as the name in the options, but just get the same  result.

Any ideas as to what to try as this looks like it would be very useful for me.

Kind regards,
Duane

DuaneLitchfield, 2018-09-26:
I have just seen one of your other responses and tried a couple of things.

First: connecting to http://192.168.1.50/setup.xml

got this
This site can’t be reached
192.168.1.50 refused to connect.
Search Google for set up
ERR_CONNECTION_REFUSED

Second: the pywemo discovery command using ssh.

I got this response...
>>> devices = pywemo.discover_devices()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pywemo/discovery.py", line 22, in discover_devices
    ssdp_entries = ssdp.scan(st, max_entries=max_devices, match_mac=match_mac)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pywemo/ssdp.py", line 245, in scan
    if entry.description is not None:
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pywemo/ssdp.py", line 140, in description
    tree = ElementTree.fromstring(xml)
  File "/usr/lib/python2.7/xml/etree/ElementTree.py", line 1311, in XML
    parser.feed(text)
  File "/usr/lib/python2.7/xml/etree/ElementTree.py", line 1651, in feed
    self._parser.Parse(data, 0)
UnicodeEncodeError: 'ascii' codec can't encode character u'\u2019' in position 357: ordinal not in range(128)
>>> 

I have tried restarting everything and re-installing the addin but no joy as yet.

Hope this helps you think of something though.

jneilliii, 2018-09-26:
Thanks for the info.  The issue definitely appears to be within the underlying python library pywemo that I use to communicate with the devices.  That error indicates it might be having an issue with a "Right Single Quotation Mark". Are you using that in any of your devices names? I'll post an issue on the pywemo repository and cross-link to this one.

DuaneLitchfield, 2018-09-26:
Hi, thanks for the quick response.
I thought exactly the same - I have several sonoff devices running Tasmota posing as wemo sockets and one genuine wemo socket.
All the devices are using basic ascii chrs in their names. I even took out the spaces.

I will try uplugging them all tomorrow and trying again to see if one of them is not playing ball.

Thanks again.

jneilliii, 2018-09-27:
@bolsoncerrado it could be. Follow the steps [here](https://github.com/jneilliii/OctoPrint-WemoSwitch/issues/1#issuecomment-420094118) and let's see what is returned.

jneilliii, 2018-09-27:
Ok @bolsoncerrado, got your post on the other issue thread.  I'll do some testing on my end with my emulated wemo devices and see if I can reproduce this issue. If you could do me a favor and try changing the names of those to not include the brackets just to see if that could be the problem or not that would be great. If it is I'll probably be able to change the code to handle the comparison during the loop through those devices.

DuaneLitchfield, 2018-09-27:
No joy for me I'm afraid.
I have tried pulling all the sonoff (tasmota) wemos and the original one. but still get the same result.
I did find that I can see th eoriginal Belkin Wemo when running **devices = pywemo.discover_devices()** as I added 
**print (entry.description.get('device', {}))** to ssdp.py 
but it clearly is finding something else that cannot be parsed.
Looking at my dhcp server, I cannot see any device names in there that have funky characters in them either.
That makes me think something is reporting back with u\'u2019' in it.
I am not very adept with Python though and that is frustrating me a little. :o(
It looks like entry is an array that is being populated with UPNP responses from devices.
**entry = UPNPEntry.from_response(response)**

Is there any way to replace the u'\u2019' chr if present whilst populating the entry array,object,list (whatever it is)?

Populating **entry** always seems to work. it is only when trying to get the description in the next line that it goes wrong.
**if entry.description is not None:**

Of course it could be way more involved than that. I know that ssdp.py is part of pywemo, but I though a bit of tinkering would help me understand a bit more of python and may help me identify a device that is causing the problem that we could let the community know.

Just rambling thoughts I guess, but I can't help trying to fix things even if I don't understand them fully.

jneilliii, 2018-09-27:
That is great help @DuaneLitchfield, and thank you again for all the leg work.  Replacing the offending character may be possible, but would require me to fork the pywemo repository, or potentially integrate it within the plugin completely.  I too am not that proficient in python, so I struggle with it sometimes too. Maybe I'll look into integrating one of the other libraries like ouimeaux (pywemo was based on this) and get better results.  It seems that repository is at least more recent.

The interesting part here is that like you I use sonoff tasmota devices for my development and the earliest release of this plugin used the `device = pywemo.discovery.device_from_description("http://192.168.1.50/setup.xml", None)` function of pywemo and worked fine but didn't work for actual wemo devices.  Can you guys try the following for me in ssh replacing the ip with your wemo's ip?

```
cd ~/oprint/bin/
./python
import pywemo
address = "192.168.1.50"
port = pywemo.ouimeaux_device.probe_wemo(address)
url = 'http://%s:%s/setup.xml' % (address, port)
url = url.replace(':None','')
device = pywemo.discovery.device_from_description(url, None)
print(device)
```

This would in theory bypass the discovery process and avoid the looping.

jneilliii, 2018-09-27:
Oh, and BTW @DuaneLitchfield, my Tasmota plugins work flawlessly both MQTT and web based variants...

DuaneLitchfield, 2018-09-28:
Good to know my Tasmota plugins should work. :o)
I just tried the code you supplied and got this response
<WeMo Switch "Printer">
Great! as my Wemo on that ip address is actually called Printer.

However, if I change it to one of my sonoff(Tasmota) devices I get
pi@octopi:~/oprint/bin $ ./python testcode.py
Traceback (most recent call last):
  File "testcode.py", line 6, in <module>
    device = pywemo.discovery.device_from_description(url, None)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pywemo/discovery.py", line 42, in device_from_description
    return device_from_uuid_and_location(uuid, mac, description_url)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pywemo/discovery.py", line 50, in device_from_uuid_and_location
    return Switch(location, mac)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pywemo/ouimeaux_device/__init__.py", line 87, in __init__
    service = Service(self, svc, base_url)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pywemo/ouimeaux_device/api/service.py", line 85, in __init__
    self._svc_config = serviceParser.parseString(xml.content).actionList
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pywemo/ouimeaux_device/api/xsd/service.py", line 1583, in parseString
    doc = parsexml_(BytesIO(inString))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pywemo/ouimeaux_device/api/xsd/service.py", line 64, in parsexml_
    doc = etree_.parse(*args, **kwargs)
  File "<string>", line 62, in parse
  File "<string>", line 38, in parse
cElementTree.ParseError: mismatched tag: line 1, column 553

not sure if that is what you expected or not, but it seesm good from the Wemo.

Thanks again for your help with this by the way.

jneilliii, 2018-09-28:
Hey quick question @DuaneLitchfield related to your wemo and that command.  Could you run the following commands and let me know what the url is reporting?

```
cd ~/oprint/bin/
./python
import pywemo
address = "192.168.1.50"
port = pywemo.ouimeaux_device.probe_wemo(address)
url = 'http://%s:%s/setup.xml' % (address, port)
url = url.replace(':None','')
print(url)
device = pywemo.discovery.device_from_description(url, None)
print(device)
```

DuaneLitchfield, 2018-09-28:
Hi @jneilliii,erer is the result of that
http://192.168.1.50:49153/setup.xml
<WeMo Switch "Printer">

jneilliii, 2018-09-28:
Awesome, now I know why my previous version wasn't working.  I had left out the port probing part and was using the default port 80 for web, which is what the Tasmota's I have respond on. I might actually revert back to that method now and that should resolve your weird character issue.

jneilliii, 2018-09-29:
@DuaneLitchfield, please try my updated version by using the following url in plugin manager and let me know how it's working out.  You'll need to switch your settings back to using ip address instead of device name.

https://github.com/jneilliii/OctoPrint-WemoSwitch/archive/0.1.1.zip

DuaneLitchfield, 2018-09-29:
Huzzah! That has worked perfectly!

closed 2018-09-30 by jneilliii