jneilliii/OctoPrint-WemoSwitch #7: Using the GCODE M80 with the IPAddress version of the plugin

totic, 2018-10-01:
Love the app, and thanks to you I can turn on/off the printer from the UI, but I am not sure how the M80 GCODE works when I am using the ip address instead of the name of the wemo device to setup the plugin

jneilliii, 2018-10-01:
It should just be `M80 192.168.0.2` making sure to use your devices ip.

totic, 2018-10-01:
@jneilliii  Tested sending `M80 10.0.1.114` from the octopi terminal and the switch didnt turn off, jut got 
```Send: M80 10.0.1.114
Recv: ok```

Any other suggestion?

totic, 2018-10-01:
Tried Turning off with M81 and IP address and it didnt work, here is the output

```
[2018-10-01 19:02:17,500] DEBUG: Received M81 command, attempting power off of 10.0.1.114.
[2018-10-01 19:02:17,503] DEBUG: {'autoConnectDelay': 10, 'autoConnect': True, 'sysCmdOn': False, 'displayWarning': True, 'ip': '10.0.1.114', 'btnColor': '#808080', 'sysRunCmdOn': '', 'sysRunCmdOff': '', 'autoDisconnect': True, 'label': 'printer', 'autoDisconnectDelay': 0, 'sysCmdOff': False, 'sysCmdOffDelay': 0, 'gcodeOffDelay': 0, 'sysCmdOnDelay': 0, 'gcodeEnabled': False, 'gcodeOnDelay': 0, 'currentState': 'off', 'warnPrinting': False, 'icon': 'icon-bolt'}
```

the wemo switch was still on, I did check http://10.0.1.114:49153/setup.xml and I was correct

jneilliii, 2018-10-01:
The gcode commands will only work if you enable the gcode option on the plugs settings.

[image]

closed 2018-10-01 by jneilliii