jneilliii/OctoPrint-WemoSwitch #8: using klipper and no M80 support

GreatBizarro, 2018-11-07:
Not an issue but a request. I use this to just turn on/off from Octoprint for now but would like to be able to turn off the printer at the end of a print. My problem is I am running klipper and it does not support the M80/M81 command. Is there another way to control the wemo that I can issue from klipper? 
Thanks

jneilliii, 2018-11-07:
I could incorporate custom @ commands to the plugin like I have for my other power related plugins that might work for you in this case. Just not sure if that causes issue with Klipper or not. Does klipper complain if you send the command `@wemoon 192.168.0.2`?

GreatBizarro, 2018-11-07:
Not sure if it will choke on that or not.
Thanks

jneilliii, 2018-11-17:
I just released version 0.1.2 that now supports the gcode action commands `@WEMOON` and `@WEMOOFF`.  If you upgrade to this version and then use the commands with the ip address of your wemo device it should resolve the klipper issue I think.

- On Command: `@WEMOON 192.168.0.2`
- Off Command: `@WEMOOFF 192.168.0.2`

closed 2018-11-17 by jneilliii