jneilliii/OctoPrint-WemoSwitch #10: Issue with GCODE automation for system shutdown

Jimboski711, 2018-11-22:
Hello great plug in.  I am having an issue where I have the plugin working via Gcode from the terminal in Octoprint but my use case is to use the M81 or @wemooff commands to automatically shut power down when a print completes.  I have tried adding both commands to CURA under End Code and in Octoprint under "When Print Completes"  I did have it work once but it was random and no longer works.  If I take the Gcode I have in my end script and copy and paste it into the terminal after a print completes it works.  The one time it worked randomly was after I added a wait command G4 P5000 to wait 5 seconds before sending the shutdown command M81 MYIP or @WEMOOFF MyIP.

Just wondering if anyone has had success with this scenario and could share the successful End Code in Cura or "After Print Completes " Code in Octoprint to shut power off after a complete print

jneilliii, 2018-11-22:
I assume your plug with the IP you are using had the gcode option enabled in plugin settings?

Jimboski711, 2018-11-22:
> This email originated from outside of the organization.

Jim Guiette - CISSP CCSP
Enterprise Security Services Leader
Information Security Services

Jimboski711, 2018-11-22:
Yes that's correct - Like I said I can turn the plug on and off from the terminal mode by hand entering in the gcode but if I add the Gcode to the print complete area in Octoprint it does not work.  Other commands do though so as an example I can turn the fans on - set a timer ect when the print completes .  It just does not see to recognize the Gcode command to power the wemo off.

jneilliii, 2018-11-22:
Ok, I'll do some testing on my end and see. Do you have any errors in octoprint.log?

MrDogbert, 2018-11-23:
Hi @jneilliii 
I'm having the same issue as @Jimboski711 ...
M81 MYIP or @WEMOOFF MyIP work fine typing them into the terminal line,
But from either Octoprint's GCode scripts "After print job completes", or putting direct at the end of the GCode file (manually, or within Cura's end script section) it does not work at all.
If it helps, I'm running Octoprint 1.3.9
GCode option is enabled in the options.

Thankyou for the Awesome plugin! Was really after a Wemo plugin!

jneilliii, 2018-11-23:
I just thought of something but haven't checked the code yet, do you guys have warn while printing enabled?

Jimboski711, 2018-11-23:
I have Warn While Printing and Warning Prompt both selected.  Also Auto Connect and Auto Disconnect with 10 and 5 seconds.    I do not have Run System Connands on.   I have not checked any log files but could if you sent me the high level instructions on how to enable them and get them.

Jim Guiette - CISSP CCSP
Enterprise Security Services Leader
Information Security Services

jneilliii, 2018-11-23:
I think if you set the gcode off delay it will resolve the problem. What is happening is the warn while printing option is blocking the gcode off command as the printer is technically still printing when it receives the instruction. This was to avoid instances where people had a delay set and started a new print. By setting the delay the printing status is not printing when that instruction is received and it should work as expected. The other option would be to uncheck the warn while printing option.

PR #17 (2019-05-05): 0.1.3
- Added thermal runaway monitoring, resolves #3.
- Changed default gcode off delay to 1 second to avoid conflicts with warn while printing option, resolves #10.
closed 2019-05-05 by jneilliii