jneilliii/OctoPrint-WemoSwitch #11: Feature Request - Power usage monitoring

WarrenSchultz, 2018-12-27:
Thanks for all your work on these plugins!

I see that the TP-Link plugin has energy monitoring on the Octoprint main screen, but the Wemo does not seem to have the same functionality. Is it possible to add that?
 Thanks!

jneilliii, 2018-12-27:
Yeah, that was a new thing I added to the TPLink as I was gifted an energy monitoring device.  I'd be glad to look into the wemo options.  Is there a specific wemo device model that includes energy monitoring? Sorry, I had developed the plugin with an emulated device and I'm not sure all the same features are available through the emulated device.

WarrenSchultz, 2018-12-27:
No worries :)

This is the device I was gifted:
Wemo Insight
https://www.amazon.com/dp/B01DBXNYCS

If you can find a way to implement support, I can assist with any testing, let me know.