jneilliii/OctoPrint-WemoSwitch #12: Feature Request: Multiple Instances of Plugin

adamnimmo, 2019-01-19:
Howdy, I really enjoy your plugin. Would it be possible to enable the ability to control multiple wemo devices from multiple OctoPrint instances within the same Pi?

jneilliii, 2019-01-19:
I understand now. I'm not sure how that would effect enabling the plugin, but let me try on my end and see what's happening.

jneilliii, 2019-01-19:
So I just went through the process of creating a second instance on my pi and am not seeing any issues with using the plugin in both.  I installed the plugin under one instance and then restarted the second octoprint instance to see the plugin loaded and then set the settings in it.  Maybe it's because I have the same plug configured in both instances that it's working for me maybe? Going to try removing the plug from the default instance and see if that breaks anything.

jneilliii, 2019-01-19:
So deleting the plug settings from the default instance made no difference for me.  I'm able to use the plugin in multiple instances running from the same pi without any problems.  Check the logs and let me know if you see anything that may be related.

adamnimmo, 2019-01-20:
Thank you for all the help, where I was breaking it was installing the plugin in both instances without restarting, somehow it was then only linking to whichever was the second install. I deleted the plugin and installed it on the primary instance, restarted and now it's working on both.

Great plugin, thanks again for the help.

jneilliii, 2019-01-20:
Yeah, the only thing that those instructions do is to add the extra config folders and set-up the services to run on startup using the separate folders and port numbers.  The actual overall octoprint installation/server is shared between all of them so if you install a plugin on one it will install on all, you just have to restart the other instances to see the change.  There are other ways to run multiple instances which would keep the octoprint server environment to be completely duplicated, which allows for maintaining separate installs etc. but this way you used seems way easier to manage.

closed 2019-01-20 by jneilliii