jneilliii/OctoPrint-WemoSwitch #13: Changing the Icon

jkkato, 2019-02-08:
When I try to change the Icon, just a blank space appears in place of the bolt icon. If I hover over the blank space, the plug label appears in the popup, and I can click on it.

I tried "dragon" (for my ender 3 printer), and "icon-dragon". I also tried "icon-biohazard" (just picked a name from the list at random). When I switch back to "icon-bolt", that works. These are all names from the linked "Cheatsheet | Font Awsome" page linked.

jneilliii, 2019-02-09:
Looks like the link got broken, it should be any of the following if I remember correctly. 

https://fontawesome.com/v3.2.1/icons/

jkkato, 2019-02-09:
That worked, Thanks.

Too bad there is no Dragon in the list. That would have matched the emblem of the printer.

closed 2019-02-09 by jkkato

jneilliii, 2019-02-09:
If you install my tab order plugin it uses the newest fontawesome icon set, but there are a couple of internal buttons that may get effected. The format is slightly different but may work.