jneilliii/OctoPrint-WemoSwitch #15: M81 disconnects Octoprint only, @WEMOOFF does nothing.

mtsgsd, 2019-02-18:
I am using Klipper and understand that there are issues using M80/81, but neither those or the @wemoon @wemooff will work. 

Gcode is enabled
After M81 192.168.1.112: There is no effect.
plugin_wemoswitch_debug.log shows:
	[2019-02-18 04:24:38,830] DEBUG: IP 192.168.1.112 is valid.
	[2019-02-18 04:24:38,830] DEBUG: Attempting to connect to 192.168.1.112
	[2019-02-18 04:25:18,172] DEBUG: Received M81 command, attempting power off of 
192.168.1.112.
	[2019-02-18 04:25:18,175] DEBUG: None

octoprint.log shows:
	2019-02-18 04:25:18,176 - octoprint.util.comm - ERROR - Error while processing hook wemoswitch for phase queuing and command M81  192.168.1.112:
 Traceback (most recent call last):
 File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 3085, in _process_command_phase
 hook_results = hook(self, phase, command, command_type, gcode, subcode=subcode, tags=tags)
		  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_wemoswitch/__init__.py", line 22                                                 0, in processGCODE
			if plug["gcodeEnabled"]:
TypeError: 'NoneType' object has no attribute '__getitem__'
		
After @WEMOOFF 192.168.1.112 Octoprint disconnects from the printer, but the switch is not triggered.
Octoprint.log shows:
    2019-02-18 04:29:43,094 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Offline"

plugin_wemoswitch_debug shows:
	[2019-02-18 04:29:44,105] DEBUG: IP 192.168.1.112 is valid.
	[2019-02-18 04:29:44,106] DEBUG: Attempting to connect to 192.168.1.112

jneilliii, 2019-02-18:
Best I can tell from the logs it appears it's not finding the switch by IP in the plugin settings and therefore gcodeenabled option is throwing an error. I'll do some testing on my end to see what might be going on here. In the meantime, can you send me a screenshot of your plug's settings?

mtsgsd, 2019-02-18:
My bad, I should have done that in the first place. IP and device name match in Wemo as well. I'll should mention that this has been a problem for a while, just not high priority, but when I first tried this I was seeing Ouimeaux errors in the log. Now there's only what you see in the OP.
[image]

jneilliii, 2019-02-19:
Ok, so I just did a test with your exact settings (with the exception of IP) and it seems to be working for me with the wemo emulated device I have to test with. This is what I get in the plugin_wemoswitch_debug.log after enabling debug logging and running the command `@WEMOOFF 192.168.0.104`.
```
[2019-02-18 23:51:59,417] DEBUG: Received @WEMOOFF command, attempting power off of 192.168.0.104.
[2019-02-18 23:51:59,417] DEBUG: {'autoConnectDelay': 10, 'autoConnect': True, 'sysCmdOn': False, 'displayWarning': False, 'ip': '192.168.0.104', 'btnColor': '#808080', 'sysRunCmdOn': '', 'sysRunCmdOff': '', 'autoDisconnect': True, 'label': 'Wemo', 'autoDisconnectDelay': '1', 'sysCmdOff': False, 'sysCmdOffDelay': 0, 'gcodeOffDelay': '3', 'sysCmdOnDelay': 0, 'gcodeEnabled': True, 'gcodeOnDelay': '1', 'currentState': 'off', 'warnPrinting': False, 'icon': 'icon-bolt'}
[2019-02-18 23:52:02,418] DEBUG: Turning off 192.168.0.104.
[2019-02-18 23:52:02,418] DEBUG: {'autoConnectDelay': 10, 'autoConnect': True, 'sysCmdOn': False, 'displayWarning': False, 'ip': '192.168.0.104', 'btnColor': '#808080', 'sysRunCmdOn': '', 'sysRunCmdOff': '', 'autoDisconnect': True, 'label': 'Wemo', 'autoDisconnectDelay': '1', 'sysCmdOff': False, 'sysCmdOffDelay': 0, 'gcodeOffDelay': '3', 'sysCmdOnDelay': 0, 'gcodeEnabled': True, 'gcodeOnDelay': '1', 'currentState': 'off', 'warnPrinting': False, 'icon': 'icon-bolt'}
[2019-02-18 23:52:03,433] DEBUG: IP 192.168.0.104 is valid.
[2019-02-18 23:52:03,433] DEBUG: Attempting to connect to 192.168.0.104
[2019-02-18 23:52:09,012] DEBUG: Getting device info from http://192.168.0.104/setup.xml
[2019-02-18 23:52:09,292] DEBUG: Found device <WeMo Switch "living_room">
[2019-02-18 23:52:09,292] DEBUG: Sending command off to 192.168.0.104
[2019-02-18 23:52:09,374] DEBUG: Checking status of 192.168.0.104.
[2019-02-18 23:52:09,374] DEBUG: IP 192.168.0.104 is valid.
[2019-02-18 23:52:09,374] DEBUG: Attempting to connect to 192.168.0.104
```
Maybe clear out the plugin's settings in config.yaml and restart OctoPrint and try starting from scratch?

mtsgsd, 2019-02-19:
I tried that, and no better. In fact I have to use the ip address for either method. Here's the entry from the config.yaml file:
~~~
  wemoswitch:
    _config_version: 1
    arrSmartplugs:
    - autoConnect: true
      autoConnectDelay: 10
      autoDisconnect: true
      autoDisconnectDelay: '3'
      btnColor: '#808080'
      currentState: unknown
      displayWarning: true
      gcodeEnabled: true
      gcodeOffDelay: '3'
      gcodeOnDelay: 0
      icon: icon-bolt
      ip: 192.168.1.112
      label: Printer
      sysCmdOff: false
      sysCmdOffDelay: 0
      sysCmdOn: false
      sysCmdOnDelay: 0
      sysRunCmdOff: ''
      sysRunCmdOn: ''
      warnPrinting: true

jneilliii, 2019-02-20:
The plugin uses pywemo which is a derivative of ouimeaux. I vaguely remember errors relative to ssdp(?) That were more warnings then anything. Nothing that would cause this to not work.

mtsgsd, 2019-02-26:
As near as I can tell, Wemo no longer allows 3rd party connections, aside from specific partners. Their SDK was discontinued 3 years ago. Access is set up thru a phone app only. I did discover that one of their partners is IFTTT, and there is an octoprint plugin for that. Got it working. I have a new switch, so maybe your software is only going to work with older models? Thanks for your help, and you can close this issue if you like.

jneilliii, 2019-02-26:
That's interesting. I'll be able to verify tonight since I went ahead and purchased an official wemo device.

jneilliii, 2019-02-27:
So I'm definitely able to control the wemo mini I got with the button in the navbar, so it's definitely still controllable using firmware version WeMo_WW_2.00.10971.PVT-OWRT-SNSV2 on that model device. I meant to ask earlier, where are you putting the @WEMOOFF command? Is it in your gcode or added in octoprint's after print complete? Also, did you ever try with the auto disconnect option unchecked?

closed 2019-03-20 by stale[bot]