jneilliii/OctoPrint-WemoSwitch #18: Ability to add device by name instead of IP

pkshima, 2019-05-11:
The plugin requires specifying the IP of the wemo switch. Would it be possible to avoid this and instead give the name (eg. '3D printer') ?

Otherwise I need to re-configure the plugin everytime the devices get a new IP.

jneilliii, 2019-05-12:
In its current implementation no, that's not possible. You could reserve the ip in your networks router shi it doesn't change though.

joshuaboniface, 2021-01-15:
FYI this may work now, as support for hostnames in PyWemo is more robust now (see https://github.com/pavoni/pywemo/pull/172).

jneilliii, 2021-01-15:
Actually looking back through my code, hostname should work, but I don't know how to test that completely since I only have one wemo device. The specific logic is [here](https://github.com/jneilliii/OctoPrint-WemoSwitch/blob/601e6255e7022b4a2a8a6aa1c928e70f7901df01/octoprint_wemoswitch/__init__.py#L564).