jneilliii/OctoPrint-WemoSwitch #19: Unable to add another switch

ilovecats516, 2019-07-18:
I currently have one Wemo switch and it works great. I'm trying to add another by going into Settings > Plugins > Wemo Switch and clicking the plus icon, but nothing happens.

jneilliii, 2019-07-18:
Can you do me a favor and open the developer console of your browser and let me know if there are any errors when you click the add button?

ilovecats516, 2019-07-18:
Yep.

`Uncaught ReferenceError: Unable to process binding "checked: function(){return thermal_runaway }"
Message: thermal_runaway is not defined
    at checked (eval at parseBindingsString (packed_libs.js?ea9fbc9a:11374), <anonymous>:3:60)
    at init (packed_libs.js?ea9fbc9a:11394)
    at packed_libs.js?ea9fbc9a:11378
    at Object.w (packed_libs.js?ea9fbc9a:11345)
    at packed_libs.js?ea9fbc9a:11377
    at Object.q (packed_libs.js?ea9fbc9a:11317)
    at m (packed_libs.js?ea9fbc9a:11377)
    at k (packed_libs.js?ea9fbc9a:11375)
    at g (packed_libs.js?ea9fbc9a:11375)
    at k (packed_libs.js?ea9fbc9a:11375)
    at g (packed_libs.js?ea9fbc9a:11375)
    at k (packed_libs.js?ea9fbc9a:11375)
    at g (packed_libs.js?ea9fbc9a:11375)
    at k (packed_libs.js?ea9fbc9a:11375)
    at g (packed_libs.js?ea9fbc9a:11375)
    at k (packed_libs.js?ea9fbc9a:11375)
    at g (packed_libs.js?ea9fbc9a:11375)
    at k (packed_libs.js?ea9fbc9a:11375)
    at g (packed_libs.js?ea9fbc9a:11375)
    at k (packed_libs.js?ea9fbc9a:11375)
    at g (packed_libs.js?ea9fbc9a:11375)
    at k (packed_libs.js?ea9fbc9a:11375)
    at g (packed_libs.js?ea9fbc9a:11375)
    at Object.a.eb (packed_libs.js?ea9fbc9a:11381)
    at a.m.i (packed_libs.js?ea9fbc9a:11314)
    at Function.Pc (packed_libs.js?ea9fbc9a:11357)
    at Function.Qc (packed_libs.js?ea9fbc9a:11357)
    at Function.aa (packed_libs.js?ea9fbc9a:11356)
    at Function.ac (packed_libs.js?ea9fbc9a:11356)
    at Function.notifySubscribers (packed_libs.js?ea9fbc9a:11343)
    at Function.fa (packed_libs.js?ea9fbc9a:11346)
    at wemoswitchViewModel.c [as selectedPlug] (packed_libs.js?ea9fbc9a:11346)
    at wemoswitchViewModel.self.addPlug (packed_plugins.js?465af643:12993)
    at HTMLAnchorElement.<anonymous> (packed_libs.js?ea9fbc9a:11395)
    at HTMLAnchorElement.dispatch (packed_libs.js?ea9fbc9a:5184)
    at HTMLAnchorElement.elemData.handle (packed_libs.js?ea9fbc9a:4992)
    at HTMLAnchorElement.sentryWrapped (packed_core.js?8c230bbf:11012)`

jneilliii, 2019-07-19:
Thanks.  Could you install an update using the URL below in plugin manager and verify it fixes the issue?

https://github.com/jneilliii/OctoPrint-WemoSwitch/archive/0.1.4.zip

ilovecats516, 2019-07-19:
Yes, that solved the problem, thank you.

https://github.com/jneilliii/OctoPrint-WemoSwitch/archive/0.1.4.zip

jneilliii, 2019-07-19:
Thanks for the confirmation. Just refused official 0.1.4 version.

closed 2019-07-19 by jneilliii