jneilliii/OctoPrint-WemoSwitch #22: Polling Wemo status doesn't seem to work as expected

cstaab, 2019-08-19:
I have a Wemo Mini connected to the plug in and everything I've tried has worked as expected with the exception of the enabling of polling status.  I anticipated that when enabled the plugin would periodically poll the on/off state of the Wemo and update the switch icon color to match.  Am I interpreting the expected operation as intended.  I have tried it using poling intervals fro 1 min to 15 min with no apparent effect.

Am I off base on the intent of the polling status option?  If not is there anything I can do to help correct this?

jneilliii, 2019-08-20:
After enabling the option I think all you have to do is refresh the page to start the polling.

cstaab, 2019-08-20:
I’ll give that a try today.

cstaab, 2019-08-20:
I've had a chance to try it and you were correct refreshing the page did
start the polling.

The only oddity I've noticed is that if I click the icon to change the Wemo
state, it changes color correctly.  If I then refresh the page (not
something I'd normally do) the color that is displayed is wrong (it is the
prior state) until after the next poll cycle.

Thanks for your assistance.
Carl

jneilliii, 2019-08-20:
I'll have to look at that. I think what is happening is the prior state is getting cached in settings, and that's the state that loads on startup.

cstaab, 2019-08-20:
Sounds good. If there is anything I can do to help let me know.

closed 2019-09-10 by stale[bot]