jneilliii/OctoPrint-WemoSwitch #24: What do GCODE On Delay/GCODE Off Delay do?

lloyb, 2019-11-23:
Hello, Sorry to sound dumb but I'm not sure what these do, right now they are both set to 20.
I've had some issues with Shutdown printer turning the plug off too soon and am wondering if theres a setting in your plugin which could be changed to have a bigger delay?
I think maybe the pi is sending the gcode to the printer and its going into memory then the pi hits the gcode to turn the switch off and kills everything before the printer has gone through its final bit of memory?

sometimes this problem happens on a small print (2 hours) and other times I print longer than that and its ok. so I'm finding it hard to know whats causing it

jneilliii, 2019-11-23:
I saw you posted on the community forum about this as well, but to answer your question, the gcode off setting should wait that amount of time before sending the off command to the plug after the command is queued. I'm not sure if the shutdown plugin would cause any issues or bypass that, but I'll double check my code this weekend to verify it's programmed the way I think it's programmed from memory.

lloyb, 2019-11-23:
Hi jneilliii, Yes I wasn't sure the best place to post this as I'm unsure if this is a wemo issue or a shutdown printer issue.
My guess is that the trigger for shutdown doesn't go to the printer and then come back to the pi to trigger but just goes from the pi. That means that it could fire before the printer finishes.
What does the On Delay do?

It might be fixed by changing the off delay to something like 120. that would be 2mins right?

jneilliii, 2019-11-23:
So I just double-checked the code and I was right.  The command does get sent after the given number of __seconds__ in the off delay.  The on delay  is the same, it wait the configured interval and the sends the on command.  Yes, if you set the delay to 120 that is the equivalent of 2 minutes.

jneilliii, 2019-11-23:
I think the issue is the command is processed during the queuing stage within OctoPrint rather than the gcode sent phase.  It would make sense to change this in a future release.

closed 2019-11-23 by jneilliii