jneilliii/OctoPrint-WemoSwitch #25: M81 command not working after job completion

anquinn, 2020-01-09:
The Wemo switch is not turning off after a job is completed.

The button in the navbar works when clicked to turn on/off the switch. Sending M81 followed by the IP address in the terminal works.

Adding M81 followed by the IP address "M81 192.168.1.129" to the "After print job completes" setting in Octoprint does not turn off the device.

Wemo settings:
[image]

jneilliii, 2020-01-09:
Try increasing the gcode off delay to something over 1 second.

anquinn, 2020-01-09:
Thats was fast. Just running a print now to test. Will report back.

anquinn, 2020-01-09:
60 second delay seems to have done it. Thanks!

closed 2020-01-09 by anquinn