jneilliii/OctoPrint-WemoSwitch #27: Off on Idle

West1967, 2020-04-29:
Are there any plans to add 'Off on Idle' like the TP-Link Smartplug has?

jneilliii, 2020-04-29:
Yes,  eventually.

West1967, 2020-04-29:
Thanks. I'm really enjoying your plugins.

cbarth3, 2020-06-30:
Any updated plans for this?

jneilliii, 2020-06-30:
Not yet. It will get done, just haven't had the time with work lately.

jneilliii, 2020-07-26:
I've got the first changes integrated in a separate branch that enables the idle timer, add startup event monitoring, and adds upload event monitoring (requires OctoPrint 1.4.1). When the pi is powered on and the statup event monitoring is enabled all wemos with that option enabled in their individual settings will be powered on. With OctoPrint 1.4.1 if the pi is already on and you upload a file, from PrusaSlicer for example, and the options to start the print on upload is enabled the wemos with that setting enabled in their individual settings will be powered on, wait for the auto connection delay, and then start the uploaded print. If you'd like to give it some testing prior to release that would be very helpful. You can just use the URL below in plugin manager > get more> ...from URL, and click install. You can install right over the existing install. 

https://github.com/jneilliii/OctoPrint-WemoSwitch/archive/0.1.6.zip

I'll probably integrate a couple of other features before I release an official update, like a font selector that I've integrated in my other plugins and get some screenshots update along with the readme. Please report back if you have any issues with the upgrade process of expected functionality.

cbarth3, 2020-07-26:
Awesome thank you for the work on this. I will get it downloaded on one of
my printers to test. Probably won't be able to test the pi powered on
feature much as mine are always on.

PR #32 (2020-07-29): 0.1.6
- added idle timer with wait for heater and abort capabilities, resolves #27 
- added on startup event monitoring, resolves #31 
- added on upload event monitoring (requires OctoPrint 1.4.1 and start printing true in api call)
- added sidebar for toggling on/off automatic power off and force status update for all wemos
- added icon picker
- upgraded to fontawesome 5 for icons
- updated readme with more accurate settings details and new screenshots, resolves #26
closed 2020-07-29 by jneilliii