jneilliii/OctoPrint-WemoSwitch #28: How shutdown raspberry pi?

darktech666, 2020-04-30:
Hi,

I'm planning to power my Pi from the print power supply and all that connected to a Wemo switch.
Is there a possibility to use "Run System Command Before Off" to shutdown properly the Pi running Octoprint... before actually have the plugin send a turn off command to the Wemo?
I guess we need 2 delays command send at the same time but 2 different delay...

Thanks,

Martin

jneilliii, 2020-04-30:
In its current implementation I don't think you can achieve this. Do you know if the wemo had any form of timer option? I'm able to do this for tplink and tasmota plugins because they have backlog and countdown timers available.

darktech666, 2020-04-30:
Honestly don't know.
I have not purchase the Wemo plug yet... but you think I could do it with just the tplink smart plug?

jneilliii, 2020-04-30:
I know it's possible with tplink and sonoff plugs/relays that have been flashed with Tasmota.

closed 2020-05-22 by stale[bot]