jneilliii/OctoPrint-WemoSwitch #29: Not waiting for "cooling" setpoint (M109 R50) before powering off anymore

alaorath, 2020-06-17:
Due to microSD card corruption, I had to rebuild my OctoPi/OctoPrint completely. I installed the latest Octoprint (1.4.0) with your Wemo plugin (0.1.5).

The plugin is configured and turns the printer on and off as expected, but my issue lies with the end gcode.  I re-wrote it from memory, but it's not working as expected:

```
M109 R50                 ; set 50C and wait for cooling - accurate
M81 192.168.xxx.xxx    ; (IP address obfuscated)
```

I've tried other tricks (like adding a `G4 S120` "dwell" command before and after the `M109`... still not the result I want - the Wemo is switched off immediately after the print finishes, it is not obeying the "cooling" command.
I have verifed that commenting out the `M81` command, the behavior is exactly as I expect (the printer setpoint is 50 degrees and status is "cooling" with "Recv: echo:busy: processing" in the terminal until 50 degrees is reached).

I'm not sure what I forgot or overlooked from replicating my configuration from 2 weeks ago where it was working perfectly...

Any assistance would be greatly appreciated.

Complete list of Plugins & versions installed...
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/backup
|  Cancel Objects (0.4.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_cancelobject
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/corewizard
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/discovery
|  Error Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/errortracking
|  FileManager (0.1.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filemanager
|  Fullscreen Plugin (0.0.5) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_fullscreen
|  GcodeEditor (0.2.8) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_GcodeEditor
|  Logging (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/logging
|  Login UI (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/loginui
|  MultiCam (0.2.6) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_multicam
|  Octolapse (0.4.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse
|  OctoPrint-IFTTT (1.2.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_IFTTT
|  PauseForUser Event Plugin (0.1.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_pause_for_user_event
|  Pi Support Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager
|  PortLister (0.1.10) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_portlister
|  Printer Safety Check (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/printer_safety_check
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate
|  Themeify (1.2.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_themeify
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/virtual_printer
|  Wemo Switch (0.1.5) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_wemoswitch

jneilliii, 2020-06-20:
I'm not really sure here. I have plans on adding the idle timer that I've merged into some of my other power plugins, but just haven't had the time. What you have there appears to be correct, but I wonder if `M109 S50` would make a difference? Also, are you entering that in OctoPrint's Gcode settings or in your slicer?

alaorath, 2020-06-21:
An Idle power-off would be useful, indeed.

This is within OctoPrint's "print finished" gcode.  I'll give `M109 S50` a try (but if I recall, it's for a "heating" setpoint, not a cooling one).

Right now I have a cludgy work-around using `G4 S450` instead... effectively triggering 7.5 minutes of "sleep" - blocking:

```M117 Starting Cooldown
M109 R50                 ; set 50C and wait for cooldown

M117 Cooling complete - turning off heaters & stepper
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed
M84 X Y E ;Disable all steppers but Z

{% if last_temperature[last_position.t]['actual'] > 60 %}
G4 S450   ; sleep 7.5 minutes
{% endif %}

M81 <wemo_ip_here>    ; power off Wemo Switch (controlling printer)```

jneilliii, 2020-06-21:
That's only for Sprinter firmware...R in Marlin means accurate, as seen by the response to the command. Not sure what accurate means though...

https://reprap.org/wiki/G-code#M109:_Set_Extruder_Temperature_and_Wait

alaorath, 2020-06-21:
30min test print in progress...

I cleaned up the code a bit (realized the stepper/bed off commands were redundant as the Slicer takes care of those).  Here's the Print Complete gcode:

```M117 Starting Cooldown
;M109 R50                 ; set 50C and wait for cooldown
M109 S50                 ; set 50C and wait for cooldown

M117 Cooling complete
M104 S0 ;Turn-off hotend

; If statement should never trigger as hotend 600C!
{% if last_temperature[last_position.t]['actual'] > 600 %}
G4 S450   ; sleep 7.5 minutes
{% endif %}

M81 <wemo_ip>    ; power off Wemo Switch (controlling printer)```

jneilliii, 2020-06-21:
Also, please share your plugin's settings for the wemo if that doesn't work.

alaorath, 2020-06-21:
Didn't work.. from the terminal, looks like Octoprint ignored the M109 S50 (possibly because it was bracketed by two `M109 S0`... one from the Slicer, another layer in my end code...?)

Terminal snippet attached: [terminal_snippet.txt](https://github.com/jneilliii/OctoPrint-WemoSwitch/files/4810053/terminal_snippet.txt)
RECV: ok stripped out, temp notifications continue until room temp reached...

Updated Print Complete code and trying another test...

```M117 Starting Cooldown
M109 R50                 ; set 50C and wait for cooldown

{% if last_temperature[last_position.t]['actual'] = 600 %}
G4 S450   ; sleep 7.5 minutes
{% endif %}

M81 192.168.0.92    ; power off Wemo Switch (controlling printer)

;should ever run... but just in case...
M104 S0 ;Turn-off hotend```

Wemo plugin settings:

<img width="410" alt="Capture" src="https://user-images.githubusercontent.com/10910847/85232451-24bfe200-b3bc-11ea-81d3-b693d8900de2.PNG">

jneilliii, 2020-07-26:
The new idle timeout might help you in your situation. Using it rather than forcing a set temp to cooldown should work. Eliminates the need for gcode commands, etc. If you want to try it out you can copy/paste the URL below in plugin manager > get more > ...from URL and click install.

https://github.com/jneilliii/OctoPrint-WemoSwitch/archive/0.1.6.zip

alaorath, 2020-08-09:
Confirmed fixed with 0.1.6  (I like the new method better as well... more intuitive than manually adding post-printing gcode).

closed 2020-08-09 by alaorath