jneilliii/OctoPrint-WemoSwitch #30: Unable to connect to HomeKit Wemo

sdaitzman, 2020-07-10:
Is Wemo Mini supported?

I got the IP address from my router, and it doesn't seem to be working at all. Any debug info I can provide?

jneilliii, 2020-07-10:
Pretty sure I was using a wemo mini for development. Any errors in octoprint.log? You could enable debug logging in the plugin's settings, restart octoprint and try again. If no success, oddthe the file plugin_wemoswitch_debug.log file from octoprint's logging section of settings.

sdaitzman, 2020-07-11:
```
[2020-07-10 15:17:01,557] DEBUG: Checking status of <IPV4 IP my router said>.
[2020-07-10 15:17:01,558] DEBUG: IP <IPV4 IP my router said>is valid.
[2020-07-10 15:17:01,559] DEBUG: Attempting to connect to <IPV4 IP my router said>
[2020-07-10 15:17:29,631] DEBUG: Getting device info from http://<IPV4 IP my router said>/setup.xml
[2020-07-10 15:22:22,656] DEBUG: Checking status of <IPV4 IP my router said>.
[2020-07-10 15:22:22,657] DEBUG: IP <IPV4 IP my router said> is valid.
[2020-07-10 15:22:22,657] DEBUG: Attempting to connect to <IPV4 IP my router said>
[2020-07-10 15:22:50,751] DEBUG: Getting device info from http://<IPV4 IP my router said>/setup.xml
[2020-07-10 15:37:15,731] DEBUG: Checking status of <IPV4 IP my router said>.
[2020-07-10 15:37:15,732] DEBUG: IP <IPV4 IP my router said> is valid.
[2020-07-10 15:37:15,732] DEBUG: Attempting to connect to <IPV4 IP my router said>
[2020-07-10 15:37:43,791] DEBUG: Getting device info from http://<IPV4 IP my router said>/setup.xml
[2020-07-11 07:30:25,103] DEBUG: Checking status of <IPV4 IP my router said>.
[2020-07-11 07:30:25,104] DEBUG: IP <IPV4 IP my router said> is valid.
[2020-07-11 07:30:25,104] DEBUG: Attempting to connect to <IPV4 IP my router said>
[2020-07-11 07:30:53,231] DEBUG: Getting device info from http://<IPV4 IP my router said>setup.xml

```

jneilliii, 2020-07-11:
So no errors is a good thing, but it looks like it's not getting the information back from the device. If you don't mind if you could SSH to your pi and run the following commands (replace the ip listed) and let me know what is returned or if there are any errors. This is the module I'm using to communicate with the wemo device.

```
~/oprint/bin/python
import pywemo
address = "192.168.0.110"
port = pywemo.ouimeaux_device.probe_wemo(address)
url = 'http://%s:%i/setup.xml' % (address, port)
device = pywemo.discovery.device_from_description(url, None)
deviceinfo = device.deviceinfo.GetInformation()
print(deviceinfo)
```

Once complete press CTRL+Z on your keyboard to exit python.

sdaitzman, 2020-07-12:
```
 ▷ ssh <PI HOSTNAME>

Welcome to fish, the friendly interactive shell
~> ~/oprint/bin/python
Python 2.7.16 (default, Apr  6 2019, 01:42:57)
[GCC 8.2.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import pywemo
>>> address = "<IPV4 IP my router said>"
>>> port = pywemo.ouimeaux_device.probe_wemo(address)
>>> url = 'http://%s:%i/setup.xml' % (address, port)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: %d format: a number is required, not NoneType
>>>
```

Hit an error at this point, which makes me think the IP is just not working. Just wondering, is your Wemo Mini set up using the pre-Homekit method? Mine arrived, I scanned a HomeKit code with an iOS device, and it paired with HomeKit. I think there's a previous revision of the device that used a different pairing method, so maybe that's the issue?

jneilliii, 2020-07-12:
Mines using the standard wemo software, I've never used homekit.

sdaitzman, 2020-07-14:
Interesting! Do you know what API/type of API is used for communication, or is that all within the library you use?

jneilliii, 2020-07-15:
I've never deep dived into that library module so unfortunately I do not. I do know that someone else recently installed the plugin and it was working, but don't think they used "HomeKit".

jneilliii, 2021-01-09:
I'm curious if the recent device discovery option allows you to see your HomeKit wemos? I recommend using the current 0.1.10rc1 version though, so if you install by copying/pasting the URL below into Plugin Manager > Get More > ...from URL and clicking install. Once installed when you go into settings, the plugin will attempt to discover devices on the network.

```
https://github.com/jneilliii/OctoPrint-WemoSwitch/archive/0.1.10rc1.zip
```

closed 2021-02-26 by github-actions[bot]