jneilliii/OctoPrint-WemoSwitch #33: Takes over 1 minute to register

hanzala-123, 2020-07-27:
The plugin is working but it takes over 1 minute to toggle the switch.

jneilliii, 2020-07-27:
Can you please share screenshots of your settings?

hanzala-123, 2020-07-28:
[image: image.png]

jneilliii, 2020-07-28:
Sorry, but images don't come through on email replies. You'll have to open github and copy/paste a screenshot directly in your comment.

dburr, 2020-08-06:
Seeing the exact same thing. Takes about a minute from hitting the lightning bolt icon to the Wemo actually responding (both turning on or off.)

Not sure if this is related or not, but the other day I launched the WeMo app on my phone and got a message something to the effect of "I found some unregistered WeMos on your network, would you like to register them to your account?" Not sure of the exact wording, because I quickly and instinctively tapped "yes" without thinking.

Screenshots of my Octoprint Wemo settings attached.
[image]
[image]

jneilliii, 2020-08-06:
Could I get you guys to enable debug logging restart OctoPrint and try it a couple of times? The go into OctoPrint's settings and grab the file plugin_wemoswitch_debug.log from the logging section and drag it here into a comment so I can evaluate a little more. Also if you don't mind please check in your developer console in your browser to see if there are any errors?

dburr, 2020-08-06:
> Could I get you guys to enable debug logging restart OctoPrint and try it a couple of times? The go into OctoPrint's settings and grab the file plugin_wemoswitch_debug.log from the logging section [...]

[console.log](https://github.com/jneilliii/OctoPrint-WemoSwitch/files/5032369/console.log)
[plugin_wemoswitch_debug.log](https://github.com/jneilliii/OctoPrint-WemoSwitch/files/5032370/plugin_wemoswitch_debug.log)

closed 2020-09-12 by stale[bot]