jneilliii/OctoPrint-WemoSwitch #36: Power off Pi not working.

brianhole47201, 2020-09-09:
New tplinks won't work because of the firmware issue, so bought WeMo for new printer.  The tplink will use the sudo shutdown now command to trigger power off on the raspberry, but that same command doesn't work on the WeMo plugin.  I saw a previous post with the issue, so thought I'd answer the question you posed that was never answered.  The WeMo app does give an option for a timer if this helps you resolve the issue.

jneilliii, 2020-09-09:
Thanks. If that is the case then I can definitely add a similar option to this plugin using countdown timers. It will just be a while before I get to it. Relative to the TP-Link firmware, there have been reports of it being able to be factory reset, and then when prompted to connect to cloud skipping that step makes the device "local only" mode and may work with the plugin.

brianhole47201, 2020-09-09:
Thanks, no rush.  I actually found an old version HS105 tplink like the one I have that works.  I'll know tomorrow when it gets here, but I'm guessing it'll work since it's older hardware, it's probably the firmware needed as well.  The new ones I got were HS103, which I think are newer.  I'm hoping the one I ordered works.  My existing one does tell me there is a firmware update available.  I didn't let it do the update, but hopefully the new one has been around long enough to not be updated.

CodeMonkeyX, 2021-02-14:
I second something like this. Also, it would be great to have a timer on the "Run system command" options for each switch. For example, I am trying to send a shutdown command via ssh to a different pi before I turn off the plug. 
But yeah being able to gracefully shutdown the OctoPi would be fantastic too.

cetemadi, 2023-04-09:
So for the Run System Command Before Off option, the expected behavior is to run the command first and power off after the delay? The readme implies that but my plug seems to be ignoring the delay timer. When I use `sudo shutdown -h now` as the system command with a 60 second delay, the plug still shutdowns immediately.

> Run System Command Before Off: run configured system command and then power off the wemo after configured delay in seconds.