jneilliii/OctoPrint-WemoSwitch #37: Wemo is not working with octoprint

akiltopu, 2020-11-02:
Hi,

I just installed WemoSwitch plugin i configured with the ip address of wemo but when i click the icon in octoprint control panel nothing happens... 

I am using macosx Catalina for octoprint. Also I am using latest firmware for Wemo Switch

My Debug File;

[2020-11-02 23:35:28,360] DEBUG: Checking status of 192.168.1.11.
[2020-11-02 23:35:28,360] DEBUG: IP 192.168.1.11 is valid.
[2020-11-02 23:35:28,360] DEBUG: Attempting to connect to 192.168.1.11
[2020-11-02 23:35:28,377] DEBUG: Getting device info from http://192.168.1.11:49153/setup.xml

My .xml file:
<root xmlns="urn:Belkin:device-1-0">
<specVersion>
<major>1</major>
<minor>0</minor>
</specVersion>
<device>
<deviceType>urn:Belkin:device:insight:1</deviceType>
<friendlyName>Ender3</friendlyName>
<manufacturer>Belkin International Inc.</manufacturer>
<manufacturerURL>http://www.belkin.com</manufacturerURL>
<modelDescription>Belkin Insight 1.0</modelDescription>
<modelName>Insight</modelName>
<modelNumber>1.0</modelNumber>
<modelURL>http://www.belkin.com/plugin/</modelURL>
<serialNumber>XXXXXXXXXXXX </serialNumber>
<UDN>uuid:Insight-1_0-221443K1200409</UDN>
<UPC>123456789</UPC>
<macAddress>XXXXXXXXXXX</macAddress>
<firmwareVersion>WeMo_WW_2.00.11483.PVT-OWRT-Insight</firmwareVersion>
<iconVersion>0|49153</iconVersion>
<binaryState>0</binaryState>
<binaryOption>1</binaryOption>
<new_algo>1</new_algo>
<iconList>
<icon>
<mimetype>jpg</mimetype>
<width>100</width>
<height>100</height>
<depth>100</depth>
<url>icon.jpg</url>
</icon>
</iconList>
<serviceList>
<service>
<serviceType>urn:Belkin:service:WiFiSetup:1</serviceType>
<serviceId>urn:Belkin:serviceId:WiFiSetup1</serviceId>
<controlURL>/upnp/control/WiFiSetup1</controlURL>
<eventSubURL>/upnp/event/WiFiSetup1</eventSubURL>
<SCPDURL>/setupservice.xml</SCPDURL>
</service>
<service>
<serviceType>urn:Belkin:service:timesync:1</serviceType>
<serviceId>urn:Belkin:serviceId:timesync1</serviceId>
<controlURL>/upnp/control/timesync1</controlURL>
<eventSubURL>/upnp/event/timesync1</eventSubURL>
<SCPDURL>/timesyncservice.xml</SCPDURL>
</service>
<service>
<serviceType>urn:Belkin:service:basicevent:1</serviceType>
<serviceId>urn:Belkin:serviceId:basicevent1</serviceId>
<controlURL>/upnp/control/basicevent1</controlURL>
<eventSubURL>/upnp/event/basicevent1</eventSubURL>
<SCPDURL>/eventservice.xml</SCPDURL>
</service>
<service>
<serviceType>urn:Belkin:service:firmwareupdate:1</serviceType>
<serviceId>urn:Belkin:serviceId:firmwareupdate1</serviceId>
<controlURL>/upnp/control/firmwareupdate1</controlURL>
<eventSubURL>/upnp/event/firmwareupdate1</eventSubURL>
<SCPDURL>/firmwareupdate.xml</SCPDURL>
</service>
<service>
<serviceType>urn:Belkin:service:rules:1</serviceType>
<serviceId>urn:Belkin:serviceId:rules1</serviceId>
<controlURL>/upnp/control/rules1</controlURL>
<eventSubURL>/upnp/event/rules1</eventSubURL>
<SCPDURL>/rulesservice.xml</SCPDURL>
</service>
<service>
<serviceType>urn:Belkin:service:metainfo:1</serviceType>
<serviceId>urn:Belkin:serviceId:metainfo1</serviceId>
<controlURL>/upnp/control/metainfo1</controlURL>
<eventSubURL>/upnp/event/metainfo1</eventSubURL>
<SCPDURL>/metainfoservice.xml</SCPDURL>
</service>
<service>
<serviceType>urn:Belkin:service:remoteaccess:1</serviceType>
<serviceId>urn:Belkin:serviceId:remoteaccess1</serviceId>
<controlURL>/upnp/control/remoteaccess1</controlURL>
<eventSubURL>/upnp/event/remoteaccess1</eventSubURL>
<SCPDURL>/remoteaccess.xml</SCPDURL>
</service>
<service>
<serviceType>urn:Belkin:service:deviceinfo:1</serviceType>
<serviceId>urn:Belkin:serviceId:deviceinfo1</serviceId>
<controlURL>/upnp/control/deviceinfo1</controlURL>
<eventSubURL>/upnp/event/deviceinfo1</eventSubURL>
<SCPDURL>/deviceinfoservice.xml</SCPDURL>
</service>
<service>
<serviceType>urn:Belkin:service:insight:1</serviceType>
<serviceId>urn:Belkin:serviceId:insight1</serviceId>
<controlURL>/upnp/control/insight1</controlURL>
<eventSubURL>/upnp/event/insight1</eventSubURL>
<SCPDURL>/insightservice.xml</SCPDURL>
</service>
<service>
<serviceType>urn:Belkin:service:smartsetup:1</serviceType>
<serviceId>urn:Belkin:serviceId:smartsetup1</serviceId>
<controlURL>/upnp/control/smartsetup1</controlURL>
<eventSubURL>/upnp/event/smartsetup1</eventSubURL>
<SCPDURL>/smartsetup.xml</SCPDURL>
</service>
<service>
<serviceType>urn:Belkin:service:manufacture:1</serviceType>
<serviceId>urn:Belkin:serviceId:manufacture1</serviceId>
<controlURL>/upnp/control/manufacture1</controlURL>
<eventSubURL>/upnp/event/manufacture1</eventSubURL>
<SCPDURL>/manufacture.xml</SCPDURL>
</service>
</serviceList>
<presentationURL>/pluginpres.html</presentationURL>
</device>
</root>

jneilliii, 2020-11-02:
I wonder if this is because the device type you are using. The pywemo module is what I use to communicate with the wemo device. Your device appears to be returning `<deviceType>urn:Belkin:device:insight:1</deviceType>` where mine that I've tested this plugin with is `<deviceType>urn:Belkin:device:controllee:1</deviceType>`. Firmware version running on my device is `WeMo_WW_2.00.10971.PVT-OWRT-SNSV2` and yours appears to be `WeMo_WW_2.00.11483.PVT-OWRT-Insight` which would make sense since those are device specific. 

The plugin uses the same strategy as described [here](https://github.com/pavoni/pywemo#if-discovery-doesnt-work-on-your-network) for establishing the connect and even the example seems to indicate that this should work with insight device. I just don't have one to test with specifically. I did just test my plugin with the wemo plug that i do have and it is still working here. Based on the log you provided it seems that the device is never discovered. It should have something like the below after that line of getting device info.

```
[2020-11-02 22:48:16,303] DEBUG: Found device <WeMo Switch "Wemo">
[2020-11-02 22:48:16,305] DEBUG: Sending command info to 192.168.0.116
```

akiltopu, 2020-11-03:
very bad for me :( 

is there anything else i can do with octoprint for power on/off my ender v2 device ?

Thanks

jneilliii, 2020-11-03:
If you can figure out how to use the pywemo module linked above with your device it might be possible to make it work, but because I don't have one specifically it's hard for me to program for it.

akiltopu, 2020-11-08:
Hi again, 

I am trying  to learn python and i install pywemo 

import pywemo

devices = pywemo.discover_devices()
print(devices)

when i run this command i am receiving

[<WeMo Insight "Ender3">]
[Finished in 6.369s]

so it discovers my plug correctly. I don't understand why your implementation is not working ? 

any guess ?

Thanks

jneilliii, 2020-11-08:
I cannot think of any reason it wouldn't work if that was able to. I'm using roughly the same commands, although I don't use the discover devices approach. I use something like this instead: https://github.com/pavoni/pywemo#if-discovery-doesnt-work-on-your-network. The below would be the equivalent if you open a python command line.

```
import pywemo
device = pywemo.discovery.device_from_description("http://192.168.1.11:49153/setup.xml", None)
print(device)
```

jneilliii, 2020-11-08:
that is assuming that the ip of your wemo is still 192.168.1.11.

akiltopu, 2020-11-08:
your command also working.. it returns

<WeMo Insight "Ender3">
[Finished in 0.492s]

Really interesting.

akiltopu, 2020-11-08:
Also i install Wswitch from appstore. This small utility is also working with my plug.

jneilliii, 2020-11-08:
This makes no sense to me then because that is the exact same commands I'm using in the plugin. Are you running these commands from the pi or from another device on the same network? The only thing I can think of that might be interfering is some form of WiFi AP isolation mode where communication is blocked between wifi connected devices. You don't see any other errors in your octoprint.log?

akiltopu, 2020-11-08:
I am using macosx catalina. I installed octoprint to this machine. The same computer where i run the scripts. But octoprint runs on phyton 3. I am running your scripts on phyton 2.7. İs it important?

I will look at my .log files tomorrow. :/

akiltopu, 2020-11-09:
hi again, 

when i run the script with python3 errors occur.
`python3 /Users/canakiltopu/Desktop/pywemo-master/main.py
Traceback (most recent call last):
  File "/Users/canakiltopu/Desktop/pywemo-master/main.py", line 3, in <module>
    devices = pywemo.discover_devices()
  File "/Users/canakiltopu/Desktop/pywemo-master/pywemo/discovery.py", line 38, in discover_devices
    device = device_from_description(
  File "/Users/canakiltopu/Desktop/pywemo-master/pywemo/discovery.py", line 59, in device_from_description
    return device_from_uuid_and_location(
  File "/Users/canakiltopu/Desktop/pywemo-master/pywemo/discovery.py", line 79, in device_from_uuid_and_location
    return Insight(url=location, mac=mac,
  File "/Users/canakiltopu/Desktop/pywemo-master/pywemo/ouimeaux_device/insight.py", line 17, in __init__
    self.update_insight_params()
  File "/Users/canakiltopu/Desktop/pywemo-master/pywemo/ouimeaux_device/insight.py", line 26, in update_insight_params
    params = self.insight.GetInsightParams().get('InsightParams')
  File "/Users/canakiltopu/Desktop/pywemo-master/pywemo/ouimeaux_device/api/service.py", line 70, in __call__
    for response_item in et.fromstring(
AttributeError: 'xml.etree.ElementTree.Element' object has no attribute 'getchildren'`

If i use python 2.7 there is no problem. Maybe this is the problem ?

akiltopu, 2020-11-09:
my octoprint.log file also says: 

`2020-11-09 12:58:37,601 - tornado.access - ERROR - 500 POST /api/plugin/wemoswitch (::1) 481.93ms
2020-11-09 12:58:38,140 - octoprint.server.util.sockjs - INFO - User root logged in on the socket from client ::1
2020-11-09 12:58:47,120 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin wemoswitch
Traceback (most recent call last):
  File "/Users/canakiltopu/OctoPrint/venv/lib/python3.9/site-packages/octoprint/server/api/__init__.py", line 109, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/Users/canakiltopu/OctoPrint/venv/lib/python3.9/site-packages/octoprint_wemoswitch/__init__.py", line 291, in on_api_command
    self.check_status("{ip}".format(**data))
  File "/Users/canakiltopu/OctoPrint/venv/lib/python3.9/site-packages/octoprint_wemoswitch/__init__.py", line 263, in check_status
    chk = self.sendCommand("info", plugip)
  File "/Users/canakiltopu/OctoPrint/venv/lib/python3.9/site-packages/octoprint_wemoswitch/__init__.py", line 554, in sendCommand
    device = pywemo.discovery.device_from_description(url, None)
  File "/Users/canakiltopu/OctoPrint/venv/lib/python3.9/site-packages/pywemo/discovery.py", line 55, in device_from_description
    return device_from_uuid_and_location(
  File "/Users/canakiltopu/OctoPrint/venv/lib/python3.9/site-packages/pywemo/discovery.py", line 75, in device_from_uuid_and_location
    return Insight(url=location, mac=mac,
  File "/Users/canakiltopu/OctoPrint/venv/lib/python3.9/site-packages/pywemo/ouimeaux_device/insight.py", line 17, in __init__
    self.update_insight_params()
  File "/Users/canakiltopu/OctoPrint/venv/lib/python3.9/site-packages/pywemo/ouimeaux_device/insight.py", line 26, in update_insight_params
    params = self.insight.GetInsightParams().get('InsightParams')
  File "/Users/canakiltopu/OctoPrint/venv/lib/python3.9/site-packages/pywemo/ouimeaux_device/api/service.py", line 70, in __call__
    for response_item in et.fromstring(
AttributeError: 'xml.etree.ElementTree.Element' object has no attribute 'getchildren'
2020-11-09 12:58:47,123 - tornado.access - ERROR - 500 POST /api/plugin/wemoswitch (::1) 256.38ms
`

Same thing, python 3.9 problem i think.. 

Can you fix this ?

jneilliii, 2020-11-09:
There we go. this may actually be a python 3.9 issue with the pywemo module. All my python 3 testing has been with 3.7.3. Let me check upstream with the pywemo project and see if there is any indication of a fix.

jneilliii, 2020-11-09:
Yep, it's an upstream error with 3.9. https://github.com/pavoni/pywemo/issues/173 there appears to be a pull request to that project in order to resolve it. Once that gets merged, I think you'll be able to upgrade the module in your virtualenv for octoprint to get this to work.

akiltopu, 2020-11-09:
for now can i downgrade to 3.7.3 my octoprint python version ?

do u know an easy way to downgrade?

jneilliii, 2020-11-09:
I do not know an easy way to downgrade python version, sorry.

jneilliii, 2021-01-09:
I see this PR got merged and the plugin should be working now in the newer versions of Python. Closing Issue.

closed 2021-01-09 by jneilliii