jneilliii/OctoPrint-WemoSwitch #41: 0.1.9 completely broke all my Wemos on octoprint

Jerry-Koh, 2021-01-08:
had 0.1.8 working perfectly fine with 2 Wemo switches on 2 separate octoprint installs on Pi4, loved that it turned off the printers on idle (especially during an overnight print), took the upgrade to 0.1.9, now nothing works, can't turn on, off, turn off on idle... uninstall and was lookign to reinstall 0.1.8, but it keeps giving me 0.1.9 :( so unhappy now...

Jerry-Koh, 2021-01-08:
figured out how to get 0.1.8 back, and it works now. Whatever was done between 0.1.8 and 0.1.9 was destructive to my set up. Let me know if you need logs (not sure where they are stored)

jneilliii, 2021-01-08:
That's strange, all my testing it was working fine. If you could grab your octoprint.log, and if you had debug logging enabled in the plugin the file plugin_wemoswitch_debug.log file from the logging section of OctoPrint that would be great. You can upload them here by dragging them into a comment.

jneilliii, 2021-01-08:
I'm able to reproduce the issue, thanks for reporting. I'll get this fixed ASAP. In the meantime I have reverted the 0.1.9 version to prerelease.

jneilliii, 2021-01-08:
Should be fixed now.

closed 2021-01-08 by jneilliii