jneilliii/OctoPrint-WemoSwitch #42: New discovery UI challange

dstrohl, 2021-01-09:
It's kind of minor, but if I have multiple "plugs" setup in the system, (at least when I have 2), both of them show the same plug in the discovered dropdown.  They are correct in terms of the data in the address field, and they do seem to update correctly when I close the box, but it made me wonder if I selected the wrong item to edit at first.

jneilliii, 2021-01-09:
The discovered dropdown is what is found on the network during OctoPrint startup. I'm already thinking of changing that around to onSettingsShown to be able to update the list when the settings are opened. I didn't realize the on_settings_load that I implemented only happens on startup. Are you saying you see two of the same item in the dropdown with the same serial number?

dstrohl, 2021-01-09:
OK, I uploaded a video of what I am seeing, here are the steps:
1. add a new switch, click "+" and the editor comes up, showing the "choose discovered device". 
2. select the device and save it. (selecting "outlet 4")
3. edit one of the other existing device, you can see that the selector still shows the "outlet 4" device, even though this a different device.
4. note that if I change variables here it does seem to correctly change things for the correct switch, so it's not overwriting the device or anything,.
5. next I go back and forth and check to see that the ip addresses and port numbers are correct even though the drop down shows a different device.
6. finally, I exit out of setup (and save) then go back in and see that now it has gone back to "select a device".

What I would expect is that it would either:
a. show the selected device each time I go back in (preferred)
b: show the "select a device" default position when I go back in

I'm not sure what the cause is here, my suspicion is that somewhere (either you or octoprint) is using the same object for each of the new switch options and not clearing the existing variables before displaying the new device dialog.

https://user-images.githubusercontent.com/9784892/104109468-187d7280-5283-11eb-91c1-c6b65bc5ee03.mp4

jneilliii, 2021-01-09:
Ah, I get it now. Yeah, this is just a little UI tweak that needs to be done. Thanks for the video, it's a little harder for me to understand without having more than the one device. I'll see what I can do to implement the preferred option...

jneilliii, 2021-01-09:
Unfortunately, I wasn't able to get you preferred method to work because of the knockout bindings I'm assuming, but was able to reset the selected option when editing/adding devices. The 0.1.10rc2 version has this now if you want to double-check.

closed 2021-01-12 by jneilliii