jneilliii/OctoPrint-WemoSwitch #43: Discovery only happens at octoprint startup.

dstrohl, 2021-01-09:
So, if I add a new device while octoprint is already running it does not show up in the discovery dropdown.  Not a big deal though something to be aware of in case someone asks why it didnt find something, but perhaps next rec it might make sense to redo the discovery on rendering the settings ui?

jneilliii, 2021-01-09:
Yeah, as mentioned in your other posted issue, this is something I plan to change.

jneilliii, 2021-01-09:
I have moved this over to onSettingsShown. If you switch the release channel in OctoPrint's software update settings for this plugin, you should get a notification to upgrade to 0.1.20rc1. Once I have verification from you I can make an official release.

dstrohl, 2021-01-09:
yup, seems to work fine.  probably worth putting a note in the docs / readme that indicates that it scans when loading the octoprint settings page (as opposed to the add device page in the plugin).  it's not a big deal, but if you don't know, I could see someone wondering.

jneilliii, 2021-01-09:
Yeah, I just pushed 0.1.10rc2 that shows a little better in the UI that is trying to discover when it's happening in the edit/add modals.

closed 2021-01-12 by jneilliii