jneilliii/OctoPrint-WemoSwitch #44: Wemo control stopped working after latest plugin update 0.1.9

mredcc, 2021-01-11:
Wemo control was working great under 0.1.8

When I was prompted to update to 0.1.9 under octopi, I couldn't get it to work anymore.  I removed 0.1.9 and installed 0.1.8 again and everything seems to be working fine so far.

Debug log under 0.1.9 didn't show much useful information, just multiple lines of:

[2021-01-11 07:28:41,226] DEBUG: Checking status of 192.168.1.212:49153.

jneilliii, 2021-01-11:
Can you please check octoprint.log for any errors related to the wemoswitch plugin? There was an initial issue #41 with the release for previously configured plugs but your checking status command indicates it was using the new discovery method with port assigned already. I re-released 0.1.9 to fix that one as it was immediately reported.

mredcc, 2021-01-11:
Here's what I'm seeing in the ocotprint.log:

2021-01-11 06:45:51,402 - octoprint.plugins.wemoswitch - INFO - WemoSwitch loaded!
2021-01-11 06:45:51,659 - octoprint.plugins.announcements - INFO - Loaded channel _octopi from https://octoprint.org/feeds/octopi.xml in 0.85s
2021-01-11 06:45:53,095 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2021-01-11 06:45:57,046 - octoprint.plugin - ERROR - Error while calling plugin wemoswitch
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_wemoswitch/__init__.py", line 119, in on_after_startup
    self._wemoswitch_logger.debug("Device %s: %s" % d)
TypeError: not enough arguments for format string

.
.
.

2021-01-11 06:56:27,399 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin wemoswitch
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 146, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_wemoswitch/__init__.py", line 323, in on_api_command
    self.check_status("{ip}".format(**data))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_wemoswitch/__init__.py", line 295, in check_status
    chk = self.sendCommand("info", plugip)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_wemoswitch/__init__.py", line 569, in sendCommand
    plugip, port = plugip.split(':', maxsplit=1)
TypeError: split() takes no keyword arguments
2021-01-11 06:56:27,411 - tornado.access - ERROR - 500 POST /api/plugin/wemoswitch (fe80::4536:9cd0:a3c7:a3af) 22.46ms
2021-01-11 06:56:49,869 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin wemoswitch
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 146, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_wemoswitch/__init__.py", line 323, in on_api_command
    self.check_status("{ip}".format(**data))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_wemoswitch/__init__.py", line 295, in check_status
    chk = self.sendCommand("info", plugip)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_wemoswitch/__init__.py", line 569, in sendCommand
    plugip, port = plugip.split(':', maxsplit=1)
TypeError: split() takes no keyword arguments
2021-01-11 06:56:49,882 - tornado.access - ERROR - 500 POST /api/plugin/wemoswitch (fe80::4536:9cd0:a3c7:a3af) 23.52ms

jneilliii, 2021-01-11:
Thanks I'll get that squared away.

jneilliii, 2021-01-11:
Can you try 0.1.10rc3 for me to see if it works for you. I have fixed the python 2 compatibility issue that was showing in your log and I think that will resolve the problem. You can go into OctoPrint's software update settings and switch the Release Channel for the plugin to Release Candidate and it should prompt you to update. That is if you are using OctoPrint 1.5.0+. If not you'll have to manually copy/paste the URL below into plugin manager > get more > ...from URL and clicking Install.

```
https://github.com/jneilliii/OctoPrint-WemoSwitch/archive/0.1.10rc3.zip
```

mredcc, 2021-01-11:
I updated the plugin from 0.1.8 to 0.1.10rc3 using the OctoPrint software update and the issue seems to be resolved.  I am able to manually control the wemo plug power and I don't see any related errors in the logs.

jneilliii, 2021-01-11:
awesome, thanks for verification. I'll release it official tonight after my work day is complete.

closed 2021-01-12 by jneilliii