jneilliii/OctoPrint-WemoSwitch #46: Feature Request - Ability to trigger status on print start and/or preheat

joshuaboniface, 2021-01-15:
It could be cool to have some more options for when to turn on the switches, especially an option for "turn on when print starts".

Context: I've got two WeMos with my system. One controls the printer, the other controls a light that provides illumination for the webcam monitoring. It would be cool to have the ability to turn on the latter automatically when a print starts, or when the bed begins preheating for a print. Could also perhaps integrate well with #11 to begin polling energy usage on heating/print start.

jneilliii, 2021-01-15:
That was kind of the reason for the gcode triggers. You can put those whereever you want, in your slicer start/end gcode or in octoprint's gcode scripts section, etc.

joshuaboniface, 2021-01-15:
That makes sense. I've also looked into hacking this in as an option and so far it's working, so if you'd accept that I'll do up a PR, otherwise I can use the GCode.

joshuaboniface, 2021-01-15:
Actually I've run into issues with config migration, so, this seems more sensible in general. Thanks!

closed 2021-01-15 by joshuaboniface

jneilliii, 2021-01-15:
Cool. No problem. For future reference, I am always open to PRs for improved codebase and additional features.

joshuaboniface, 2021-01-15:
Great to know! What I added wasn't particularly interesting or much of an improvement (just a duplication of the "on_upload" functionality), so to save the hassle of having to migrate configs I'll stick with the GCode method.