jneilliii/OctoPrint-WemoSwitch #48: OctoPrint Install Fails- Identified as "Unknown" by PluginManager

raffenit, 2021-03-05:
First and subsequent attempts at installing result in no plugin installation in OctoPrint. Not sure if a version issue, but I was unable to determine the cause.

Install log looks fine on each install attempt, but:

1. Plugin Manager notification shows "Unknown" as the plugin name
2. On server restart, plugin is not listed in Plugin Manager, and is included in the list of installable plugins
3. When we click "Install" again, this process repeats

Install Log from subsequent installs:
```
Installing plugin "OctoPrint-WemoSwitch" from https://github.com/jneilliii/OctoPrint-WemoSwitch/archive/master.zip...
/home/pi/oprint/bin/python -m pip --disable-pip-version-check install file:///tmp/tmpcn77ge1k/OctoPrint-WemoSwitch-master.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmpcn77ge1k/OctoPrint-WemoSwitch-master.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-WemoSwitch==0.1.10) (1.5.3)
Requirement already satisfied: pywemo in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-WemoSwitch==0.1.10) (0.6.3)
Requirement already satisfied: netifaces<1,>=0.10.9 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (0.10.9)
Requirement already satisfied: netaddr<1,>=0.7.19 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (0.8.0)
Requirement already satisfied: watchdog==0.10.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (0.10.4)
Requirement already satisfied: pyserial<4,>=3.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (3.5)
Requirement already satisfied: psutil<6,>=5.7 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (5.8.0)
Requirement already satisfied: pylru<2,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (1.2.0)
Requirement already satisfied: zeroconf<0.25,>=0.24 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (0.24.5)
Requirement already satisfied: filetype<2,>=1.0.7 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (1.0.7)
Requirement already satisfied: future<1,>=0.18.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (0.18.2)
Requirement already satisfied: sentry-sdk<1,>=0.15.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (0.19.5)
Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (3.1.1)
Requirement already satisfied: rsa==4.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (4.0)
Requirement already satisfied: feedparser<7,>=6.0.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (6.0.2)
Requirement already satisfied: PyYAML<6,>=5.3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (5.4.1)
Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (2.0)
Requirement already satisfied: pkginfo<2,>=1.5.0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (1.7.0)
Requirement already satisfied: frozendict<2,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (1.2)
Requirement already satisfied: OctoPrint-FileCheck>=2020.08.07 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (2021.2.23)
Requirement already satisfied: Jinja2<3,>=2.11.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (2.11.2)
Requirement already satisfied: wrapt<2,>=1.12.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (1.12.1)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2020.09.23 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (2021.2.4)
Requirement already satisfied: itsdangerous<2,>=1.1.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (1.1.0)
Requirement already satisfied: flask<2,>=1.1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (1.1.2)
Requirement already satisfied: tornado==5.1.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (5.1.1)
Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (2020.11.13)
Requirement already satisfied: werkzeug<2,>=1.0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (1.0.1)
Requirement already satisfied: semantic-version<3,>=2.8.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (2.8.5)
Requirement already satisfied: markupsafe<2.0,>=1.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (1.1.1)
Requirement already satisfied: requests<3,>=2.23.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (2.25.1)
Requirement already satisfied: websocket-client<1,>=0.57 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (0.57.0)
Requirement already satisfied: unidecode<0.05,>=0.04.14 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (0.4.21)
Requirement already satisfied: emoji<1,>=0.5.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (0.6.0)
Requirement already satisfied: blinker<2,>=1.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (1.4)
Requirement already satisfied: sarge==0.1.5post0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (0.1.5.post0)
Requirement already satisfied: Click<8,>=7.1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (7.1.2)
Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (0.5.0)
Requirement already satisfied: cachelib<1,>=0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (0.1.1)
Requirement already satisfied: Flask-Babel<2,>=1.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (1.0.0)
Requirement already satisfied: pyasn1>=0.1.3 in /home/pi/oprint/lib/python3.7/site-packages (from rsa==4.0->OctoPrint->OctoPrint-WemoSwitch==0.1.10) (0.4.8)
Requirement already satisfied: pathtools>=0.1.1 in /home/pi/oprint/lib/python3.7/site-packages (from watchdog==0.10.4->OctoPrint->OctoPrint-WemoSwitch==0.1.10) (0.1.2)
Requirement already satisfied: sgmllib3k in /home/pi/oprint/lib/python3.7/site-packages (from feedparser<7,>=6.0.2->OctoPrint->OctoPrint-WemoSwitch==0.1.10) (1.0.0)
Requirement already satisfied: webassets>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from Flask-Assets<3,>=2.0->OctoPrint->OctoPrint-WemoSwitch==0.1.10) (2.0)
Requirement already satisfied: pytz in /home/pi/oprint/lib/python3.7/site-packages (from Flask-Babel<2,>=1.0->OctoPrint->OctoPrint-WemoSwitch==0.1.10) (2020.5)
Requirement already satisfied: Babel>=2.3 in /home/pi/oprint/lib/python3.7/site-packages (from Flask-Babel<2,>=1.0->OctoPrint->OctoPrint-WemoSwitch==0.1.10) (2.9.0)
Requirement already satisfied: setuptools>=36 in /home/pi/oprint/lib/python3.7/site-packages (from markdown<3.2,>=3.1->OctoPrint->OctoPrint-WemoSwitch==0.1.10) (52.0.0)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in /home/pi/oprint/lib/python3.7/site-packages (from requests<3,>=2.23.0->OctoPrint->OctoPrint-WemoSwitch==0.1.10) (1.26.3)
Requirement already satisfied: idna<3,>=2.5 in /home/pi/oprint/lib/python3.7/site-packages (from requests<3,>=2.23.0->OctoPrint->OctoPrint-WemoSwitch==0.1.10) (2.10)
[... cut ...]
Created wheel for OctoPrint-WemoSwitch: filename=OctoPrint_WemoSwitch-0.1.10-py3-none-any.whl size=298344 sha256=655be1918397fa93c6901780e136d7692d0475fd63190c14f49757fdff0557d5
Stored in directory: /tmp/pip-ephem-wheel-cache-g1hayoww/wheels/21/36/f0/d2e625b7affcce9ae24e13dd01b1cb761fee4b85aa2b8e58c7
Successfully built OctoPrint-WemoSwitch
Installing collected packages: OctoPrint-WemoSwitch
Attempting uninstall: OctoPrint-WemoSwitch
Found existing installation: OctoPrint-WemoSwitch 0.1.10
Uninstalling OctoPrint-WemoSwitch-0.1.10:
Successfully uninstalled OctoPrint-WemoSwitch-0.1.10
Successfully installed OctoPrint-WemoSwitch-0.1.10
Done!

```
[image]

jneilliii, 2021-03-05:
Can you check your octoprint.log for errors relative to the plugin? There's nothing in that log that indicates what might be going on.

raffenit, 2021-03-05:
Found one:
`2021-03-05 05:31:02,683 - octoprint.plugin.core - ERROR - Error loading plugin wemoswitch
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 1298, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 69, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 238, in load_module
    return load_package(name, filename)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 212, in load_package
    return _load(spec)
  File "<frozen importlib._bootstrap>", line 696, in _load
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_wemoswitch/__init__.py", line 16, in <module>
    import pywemo
  File "/home/pi/oprint/lib/python3.7/site-packages/pywemo/__init__.py", line 4, in <module>
    from .discovery import discover_devices, setup_url_for_address
  File "/home/pi/oprint/lib/python3.7/site-packages/pywemo/discovery.py", line 9, in <module>
    from . import ssdp
  File "/home/pi/oprint/lib/python3.7/site-packages/pywemo/ssdp.py", line 10, in <module>
    from lxml import etree as et
ImportError: libxslt.so.1: cannot open shared object file: No such file or directory`

raffenit, 2021-03-05:
Also good to know that octoprint.log exists, it seems quite useful :)

raffenit, 2021-03-05:
Perhaps something to do with python version? That might make some sense. I updated that a few weeks ago.

jneilliii, 2021-03-05:
Can you SSH to your pi and attempt the command `sudo apt install libxslt-dev` and then restart OctoPrint and see if that brings it back to life? It appears it might be a missing library required for the pywemo dependency I use.

raffenit, 2021-03-05:
Awesome, that worked! Thank you! 😊

Strange that it wasn't already installed, is that a standard library?

jneilliii, 2021-03-05:
I'm not 100% sure. Maybe the older octopi image version 0.17 didn't have it. Going to pin this issue for others to find in case it happens to them too.

closed 2021-03-05 by jneilliii

tomatac, 2023-12-10:
This is still an issue but the fix above worked. Thanks!

theOrneryMaker, 2025-04-24:
This is broken again with version 0.1.14 installing on Octoprint 1.11.0.  The libxslt-dev solution did not resolve, it now hangs indefinitely at "Building wheel for lxml (pyproject.toml): started"

jneilliii, 2025-04-24:
@theOrneryMaker please open new issue with octoprint.log attached.