jneilliii/OctoPrint-WemoSwitch #49: Add feature to enable power on before starting print

David-Maisonave, 2021-04-22:
Can a new option be added to enable power on before starting a print job?

I've been using this plugin for about 6 months now, and I've only figured out today that I'm able to have the printer turn on automatically before starting a print job by enabling the GCODE Trigger and adding the M80 command to the GCODE script [Before print job starts].

I think many users want this functionality, but don't know how to set it up.  IMHO, it would help these users if there was an option to do this programmatically.

Another option would be to add clear instructions next to the GCODE Trigger option on how to configure for automatic power on before starting a print job.

jneilliii, 2021-04-23:
> Also if the workaround is in place, Wemo option "On on Upload" is disabled, and Cura-OctoPrint "Turn on printer" is de-configured, the printer turns on when starting a print job from Cura.

This is interesting to me. I didn't even catch the fact that your workaround bypasses the printer being disconnected? I didn't think you could start a print in that state, and therefore a beforePrintStart gcode wouldn't be possible either.

> I have the latest version of Cura installed

I haven't been able to verify this version yet (assuming 4.9), I installed but it's flaking out big time and I think it may be plugin related.

David-Maisonave, 2021-04-24:
I have Cura 4.9 installed.
The OctoPrint GUI doesn't let you start a print job when disconnected, but surprisingly you can start a print job via Cura-OctoPrint while OctoPrint is disconnected.
It would be nice if a plugin was created with option to let OctoPrint start the print job in a disconnected state.

FYI:
There is a minor issue I'm seeing with the Wemo plugin's power timeout.  I see the timeout getting displayed a second time before the previous timeout has closed or timed out.  I think it only happens when the timeout value is longer than the idle timeout.
Do you want me to create a new issue with more details?

jneilliii, 2021-04-25:
> Do you want me to create a new issue with more details?

Yes please, that's more a bug than feature enhancement.

David-Maisonave, 2022-01-19:
Work around solves this issue. See #57

closed 2022-01-19 by David-Maisonave