jneilliii/OctoPrint-WemoSwitch #51: Octoprint install error

yashm, 2021-05-18:
When installing the plugin from the "plugin Installer" menu in octoprint, I get this dialog box:
[image]

And subsequently, the plugin doesn't show up in the plugins list.

Then, after re-checking the plugins manager, the plugin shows an "install" button next to it , as if it was never installed.

OctoPrint 1.6.1
Python 3.7.3
OctoPi 0.17.0

jneilliii, 2021-05-18:
Can you please provide your octoprint.log. I'm afraid this may have something to do with an upstream dependency and python 3 requirements, but need to see the log first.

deftdawg, 2021-05-23:
This is the error from `~/octoprint/logs/octoprint.log`
```
2021-05-23 04:38:38,826 - octoprint.plugin.core - ERROR - Error loading plugin wemoswitch
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 1264, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 69, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 238, in load_module
    return load_package(name, filename)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 212, in load_package
    return _load(spec)
  File "<frozen importlib._bootstrap>", line 696, in _load
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_wemoswitch/__init__.py", line 16, in <module>
    import pywemo
  File "/home/pi/oprint/lib/python3.7/site-packages/pywemo/__init__.py", line 4, in <module>
    from .discovery import discover_devices, setup_url_for_address
  File "/home/pi/oprint/lib/python3.7/site-packages/pywemo/discovery.py", line 9, in <module>
    from . import ssdp
  File "/home/pi/oprint/lib/python3.7/site-packages/pywemo/ssdp.py", line 10, in <module>
    from lxml import etree as et
ImportError: libxslt.so.1: cannot open shared object file: No such file or directory
```

jneilliii, 2021-05-23:
Thanks for the update, to restart OP from CLI the command is `sudo service octoprint restart`.

closed 2021-05-23 by jneilliii