jneilliii/OctoPrint-WemoSwitch #53: Can't control switch via label or IP without port via GCODE

dealdiane, 2021-10-11:
I can't control my Wemo Switch via M80/81 or WEMOON/WEMOOFF GCODE commands using the label or IP parameter. It works if I use the IP with the port though `M80 <1.1.1.1:49153>`. However, port 49153 is just one of the ports that a Wemo device uses to communicate so I can't rely on this reliably. Any help appreciated. 

Here's the log from the commands.

```
[2021-10-11 10:18:26,528] DEBUG: Received M80 command, attempting power on of PRINTER_LIGHTS.
[2021-10-11 10:18:26,530] DEBUG: None
[2021-10-11 10:18:34,780] DEBUG: Received M80 command, attempting power on of 192.168.68.66.
[2021-10-11 10:18:34,781] DEBUG: None
[2021-10-11 10:18:55,898] DEBUG: Received M80 command, attempting power on of 192.168.68.66:49153.
[2021-10-11 10:18:55,900] DEBUG: {'autoConnect': False, 'autoConnectDelay': 10, 'autoDisconnect': False, 'autoDisconnectDelay': 0, 'automaticShutdownEnabled': False, 'btnColor': '#808080', 'currentState': 'off', 'displayWarning': False, 'event_on_startup': False, 'event_on_upload': False, 'gcodeEnabled': True, 'gcodeOffDelay': '0', 'gcodeOnDelay': '0', 'icon': 'far fa-lightbulb', 'ip': '192.168.68.66:49153', 'label': 'printer_lights', 'sysCmdOff': False, 'sysCmdOffDelay': 0, 'sysCmdOn': False, 'sysCmdOnDelay': 0, 'sysRunCmdOff': '', 'sysRunCmdOn': '', 'thermal_runaway': False, 'warnPrinting': False}
[2021-10-11 10:18:55,901] DEBUG: Turning on 192.168.68.66:49153.
[2021-10-11 10:18:55,907] DEBUG: {'autoConnect': False, 'autoConnectDelay': 10, 'autoDisconnect': False, 'autoDisconnectDelay': 0, 'automaticShutdownEnabled': False, 'btnColor': '#808080', 'currentState': 'off', 'displayWarning': False, 'event_on_startup': False, 'event_on_upload': False, 'gcodeEnabled': True, 'gcodeOffDelay': '0', 'gcodeOnDelay': '0', 'icon': 'far fa-lightbulb', 'ip': '192.168.68.66:49153', 'label': 'printer_lights', 'sysCmdOff': False, 'sysCmdOffDelay': 0, 'sysCmdOn': False, 'sysCmdOnDelay': 0, 'sysRunCmdOff': '', 'sysRunCmdOn': '', 'thermal_runaway': False, 'warnPrinting': False}
[2021-10-11 10:18:55,914] DEBUG: IP 192.168.68.66 is valid.
[2021-10-11 10:18:55,922] DEBUG: Attempting to connect to 192.168.68.66
[2021-10-11 10:18:55,925] DEBUG: Getting device info from http://192.168.68.66:49153/setup.xml
[2021-10-11 10:18:56,420] DEBUG: Found device <WeMo Switch "Printer Lights">
[2021-10-11 10:18:56,420] DEBUG: Sending command on to 192.168.68.66
[2021-10-11 10:18:56,469] DEBUG: Checking status of 192.168.68.66:49153.
[2021-10-11 10:18:56,470] DEBUG: IP 192.168.68.66 is valid.
[2021-10-11 10:18:56,470] DEBUG: Attempting to connect to 192.168.68.66
[2021-10-11 10:18:56,470] DEBUG: Getting device info from http://192.168.68.66:49153/setup.xml
[2021-10-11 10:18:56,893] DEBUG: Found device <WeMo Switch "Printer Lights">
[2021-10-11 10:18:56,894] DEBUG: Sending command info to 192.168.68.66
```

jneilliii, 2021-10-11:
Does the button in the navbar work without the port, so it's only a gcode issue?

jneilliii, 2021-10-11:
Interesting, I haven't worked with enough of these various devices to really know the options. I'm using a third part module to discover/control these [here](https://github.com/jneilliii/OctoPrint-WemoSwitch/blob/3ea71d0497eb7b254a2df650830994397d5e9dd4/octoprint_wemoswitch/__init__.py#L576) to try and deduce the port number, so I wonder if something in that module isn't working as expected. When you look in your settings does the IP field have the port listed? The plugin is using that field to do a lookup of the device, and if they don't match then it will never actually attempt I don't think.

dealdiane, 2021-10-11:
Ahh I see. It tries various ports until it finds one that works `PROBE_PORTS = (49153, 49152, 49154, 49151, 49155, 49156, 49157, 49158, 49159)`. This is a poor mans way of doing it (not your fault) albeit should work. Ideally, the correct port should be "discovered" through SSDP.

All that said, you got me to the right direction. The port was in the settings and once I removed that, it now falls back to probing it again so `M80/M81 <IP Address WITHOUT port>` now works. This will work for me since I have the IP reserved for that device.

I still can't get the GCODES to work with the label however and I can't see that label settings being used to look up anything in the code (didn't try hard though).

dealdiane, 2021-10-12:
Good question! Not in the github docs but in the [OctoPrint docs page](https://plugins.octoprint.org/plugins/wemoswitch/)

> GCODE
When checked this will enable the processing of M80 and M81 commands from gcode to power on/off plug. Syntax for gcode command is M80/M81 followed by device name. For example if your plug is living_room your gcode command would be M80 living_room

jneilliii, 2021-10-12:
Oh shoot. I need to fix that. It actually came across from the TasmotaMQTT plugin, because living_room would be an MQTT topic. Thanks for letting me know

closed 2021-10-12 by jneilliii