jneilliii/OctoPrint-WemoSwitch #54: Bug when Octoprint was booted

jefflorenzini, 2022-01-01:
I'll turn on my raspberry pi to boot Octoprint, and once it's all started up, the Wemo Switch code immediately wants to turn off the plug because it's been more than the idle time out since Octoprint was last running.

Can you check when Octoprint was started, and if it's been less than the idle timeout, reset the idle timeout so it doesn't immediately try and turn off the printer?

jneilliii, 2022-01-01:
That's something that I think I've fixed in my other power plugins. Seems to have something to do with the pi's time clock not being super accurate or something. I can definitely add the same logic to this plugin.

jneilliii, 2022-01-01:
Added in the above commit and available in version 0.1.11rc1. If you want to test it out switch the plugin's release channel in OctoPrint's Software Update settings to `Release Candidate` and upgrade when prompted.

jefflorenzini, 2022-01-01:
That was amazingly fast. I'll try it out, thanks!

PR #66 (2022-10-15): 0.1.11
- replace deprecated getBoolen with get_boolean
- remove duplicated bundling of fontawesome 5 if running on OctoPrint 1.5.0+
- add uptime to prevent instant idle timeout with on start option, resolves #54
closed 2022-10-15 by jneilliii