jneilliii/OctoPrint-WemoSwitch #57: [Solution] Windows OctoPrint automatically turning on power-on, before starting a print job!

David-Maisonave, 2022-01-19:
### Problem (Issue):
I started a print job today for the first time in 7 months, and I immediately ran into this issue again with printer error (heat overrun) which was triggered because the printer was off.
This is an easy mistake to make with a Windows OctoPrint setup where the 3D printer is connected via USB cable, because the USB cable powers the board on the 3D printer, and you see a display on the 3D printer as if it was turned on.
Moreover, even if you immediately notice the printer is off, and you click cancel, either it freezes or the job gets canceled but you still get the heat overrun error.

### Solution:
Add a PrintStarted event using the [OctoPrint Settings]->[Event Manager].
Populate the Command field with the following:**@WEMOON [IP-Address]:[Port No.]**
Example:
@WEMOON 10.80.80.123:49153
Note: Port number must be included with the Wemo specific IP address.
[image]
Set event to PrintStarted, set the type to gcode, and save the settings.

### Very important!!! Make sure to restart OctoPrint before testing.
[image]

And that's it.  Now when a print job is started, and the printer is off, the Wemo plugin will turn on the printer automatically.

Note:
As above snapshot shows, a PrintResumed event can also be added to make sure the printer is turned on when resuming a print job.
The screenshot has an extra event "PrintFailed", to turn off the printer when a print failure occurs. I've not tested it, and it's not required for the print start solution.

Since this solution is easy to implement, I'm closing my previous feature request to address this issue. (#49)

David-Maisonave, 2022-01-19:
FYI:
If you like to keep the printer turn off as much as possible, and you're using above solution, then I recommend disabling the **[On On Startup]** WemoSwitch option, and enabling the **[Power Off On Idle]** option.
[image]

Don't forget to also enable **[Power Off On Idle]** on the main WemoSwitch menu.
[image]

jneilliii, 2022-01-19:
Thanks for the information, but you should really fix the backpowering issue. More details can be found [here](https://community.octoprint.org/t/printers-known-to-have-backpowering-issues/25913?u=jneilliii).

closed 2022-01-19 by jneilliii

David-Maisonave, 2022-01-21:
That issue is related to Raspberry Pi.  I'm not using a Raspberry Pi.
Moreover, I don't know why someone would install OctoPrint on Windows if they're using a Raspberry Pi.
If you have a Raspberry Pi, you're better off installing OctoPrint on the Raspberry Pi itself.

The solution I posted is specific to users running OctoPrint on a **Windows PC**, and connecting to the 3D printer via USB.

jneilliii, 2022-01-21:
Oh really? My understanding that the backpowering can cause issues with serial communication in general, not just specific to pi, but if that's not the case then that's awesome.

David-Maisonave, 2022-01-21:
On every backpowering link I've read it always refers to the Raspberry Pi with a USB connection.  IMHO, the Raspberry Pi adds an extra layer of complexity which trigger these types of issues.

I've never had a backpowering serial communication problem, and I use to keep my 3D printer board connected and powered on via USB 24/7.
A few months back, I bought a [USB switch](https://smile.amazon.com/gp/product/B07T9BRNHW/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1), which now allows me to turn power off to the board without disconnecting the USB cable.  But I only use it because I don't like wasting electricity or seeing something on that I'm not using.

I would like to add a smart [switch circuit board ](https://smile.amazon.com/Inching-Self-locking-Wifi-Switch-Module/dp/B08F7M82NB/ref=sr_1_23) on the USB switch, but that would mean trying to find another OctoPrint plugin to control that switch.

Have you considered making this plugin more general, so that it can handle other smart switches?
Or have you considered creating a Fork to create a second general purpose smart switch plugin?

IMHO, this plugin has a really good interface for doing a general purpose smart switch plugin.

jneilliii, 2022-01-21:
Well, the Tasmota plugin is about as general as you can get and works for many devices, just takes the know-how to flash firmware. The domoticz plugin I made is more generalized, but requires an infrastructure in place. Outside of that you may want to look into PSU Control and it's plethora of available sub-plugins that can control various things.

David-Maisonave, 2022-01-22:
> Well, the Tasmota plugin is about as general as you can get and works for many devices, just takes the know-how to flash firmware. 

Nice idea, but you can't flash a [Wemo device with Tasmota](https://www.reddit.com/r/homeassistant/comments/e8hxqz/can_tasmota_or_esphome_be_flashed_onto_a_wemo/).

> The domoticz plugin I made is more generalized, but requires an infrastructure in place. 

It's strange you mention Domoticz, because a few days ago, I installed Domoticz on my computer, but I found it overly complicated.
I thought it would by default support popular well known devices. It does not.  The Honeywell Thermostat is the only device I recognized in the drop-down hardware list.
To try to add my Wemo switch, I first tried the [Domoticz-WEMO-Plugin](https://github.com/guino/Domoticz-WEMO). Much to my surprise Domoticz doesn't have a menu option to add plugins. I had to add the plugin via command line. I followed the plugin instructions, but after restarting Domotcz, Wemo still did not appear in the [**Type**] drop-down list.
I did find this link: [WeMo in Domoticz](https://www.domoticz.com/wiki/Wemo).  But that looks like too much work just to get a single device added.
It's ironic that Domoticz is designed for automation, but it lacks any automation for adding hardware or plugin's.
I do not recommend Domoticz to use with Wemo.  IMHO, Domoticz interface is poorly designed, and it's not ready for prime time.