jneilliii/OctoPrint-WemoSwitch #58: Installation not working

mdubash, 2022-01-20:
Since upgrading Octoprint to python 3 from 2, WemoSwitch has disappeared fom the plug-in list. I've reinstalled manually at the CLI and through the UI but, although it appears to install without error, it does not appear on the UI at all, either in the plug-in list nor on the menu bar.

I don't see any errors in the log:

`2022-01-20 18:58:18,118   /home/pi/oprint/bin/python -m pip --disable-pip-version-check install file:///tmp/tmpuqizwfrt/OctoPrint-WemoSwitch-master.zip --no-cache-dir
2022-01-20 18:58:21,156 > Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
2022-01-20 18:58:21,157 > Processing /tmp/tmpuqizwfrt/OctoPrint-WemoSwitch-master.zip
2022-01-20 18:58:21,168 > Preparing metadata (setup.py): started
2022-01-20 18:58:22,293 > Preparing metadata (setup.py): finished with status 'done'
2022-01-20 18:58:22,295 > Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-WemoSwitch==0.1.10) (1.7.3)
2022-01-20 18:58:22,296 > Requirement already satisfied: pywemo in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-WemoSwitch==0.1.10) (0.7.0)
2022-01-20 18:58:23,617 > Requirement already satisfied: werkzeug<2,>=1.0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (1.0.1)
2022-01-20 18:58:23,619 > Requirement already satisfied: future<1,>=0.18.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (0.18.2)
2022-01-20 18:58:23,621 > Requirement already satisfied: markupsafe<2.0,>=1.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (1.1.1)
2022-01-20 18:58:23,623 > Requirement already satisfied: colorlog<6,>=5.0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (5.0.1)
2022-01-20 18:58:23,623 > Requirement already satisfied: requests<3,>=2.26.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (2.27.1)
2022-01-20 18:58:23,624 > Requirement already satisfied: sentry-sdk<2,>=1.3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (1.5.1)
2022-01-20 18:58:23,624 > Requirement already satisfied: cachelib<0.2,>=0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (0.1.1)
2022-01-20 18:58:23,625 > Requirement already satisfied: emoji<2,>=1.4.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (1.6.1)
2022-01-20 18:58:23,625 > Requirement already satisfied: Flask-Babel<2,>=1.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (1.0.0)
2022-01-20 18:58:23,625 > Requirement already satisfied: itsdangerous<2,>=1.1.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (1.1.0)
2022-01-20 18:58:23,625 > Requirement already satisfied: immutabledict<3,>=2.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (2.2.1)
2022-01-20 18:58:23,626 > Requirement already satisfied: pkginfo<2,>=1.7.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (1.8.2)
2022-01-20 18:58:23,626 > Requirement already satisfied: unidecode in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (1.3.2)
2022-01-20 18:58:23,626 > Requirement already satisfied: OctoPrint-FirmwareCheck>=2021.10.11 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (2021.10.11)
2022-01-20 18:58:23,627 > Requirement already satisfied: sarge==0.1.6 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (0.1.6)
2022-01-20 18:58:23,627 > Requirement already satisfied: flask<2,>=1.1.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (1.1.4)
2022-01-20 18:58:23,627 > Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (3.1.1)
2022-01-20 18:58:23,628 > Requirement already satisfied: zeroconf<0.34,>=0.33 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (0.33.4)
2022-01-20 18:58:23,628 > Requirement already satisfied: semantic-version<3,>=2.8.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (2.8.5)
2022-01-20 18:58:23,629 > Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (2.0)
2022-01-20 18:58:23,629 > Requirement already satisfied: pathvalidate<3,>=2.4.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (2.5.0)
2022-01-20 18:58:23,629 > Requirement already satisfied: OctoPrint-PiSupport>=2021.10.28 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (2021.10.28)
2022-01-20 18:58:23,630 > Requirement already satisfied: filetype<2,>=1.0.7 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (1.0.9)
2022-01-20 18:58:23,630 > Requirement already satisfied: websocket-client<1,>=0.59 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (0.59.0)
2022-01-20 18:58:23,630 > Requirement already satisfied: Click<8,>=7.1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (7.1.2)
2022-01-20 18:58:23,630 > Requirement already satisfied: netaddr<0.9,>=0.8 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (0.8.0)
2022-01-20 18:58:23,631 > Requirement already satisfied: pylru<2,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (1.2.0)
2022-01-20 18:58:23,631 > Requirement already satisfied: watchdog==0.10.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (0.10.4)
2022-01-20 18:58:23,632 > Requirement already satisfied: Jinja2<3,>=2.11.3 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (2.11.3)
2022-01-20 18:58:23,632 > Requirement already satisfied: regex in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (2021.11.10)
2022-01-20 18:58:23,632 > Requirement already satisfied: tornado<7,>=6 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (6.1)
2022-01-20 18:58:23,633 > Requirement already satisfied: netifaces<1,>=0.11 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (0.11.0)
2022-01-20 18:58:23,633 > Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (0.5.0)
2022-01-20 18:58:23,633 > Requirement already satisfied: PyYAML<6,>=5.4.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (5.4.1)
2022-01-20 18:58:23,634 > Requirement already satisfied: psutil<6,>=5.8 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (5.9.0)
2022-01-20 18:58:23,634 > Requirement already satisfied: feedparser<7,>=6.0.8 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (6.0.8)
2022-01-20 18:58:23,634 > Requirement already satisfied: OctoPrint-FileCheck>=2021.2.23 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (2021.2.23)
2022-01-20 18:58:23,635 > Requirement already satisfied: blinker<2,>=1.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (1.4)
2022-01-20 18:58:23,635 > Requirement already satisfied: wrapt<1.13,>=1.12.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (1.12.1)
2022-01-20 18:58:23,635 > Requirement already satisfied: pyserial<4,>=3.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (3.5)
2022-01-20 18:58:23,636 > Requirement already satisfied: zipstream-new<1.2,>=1.1.8 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.10) (1.1.8)
[... cut ...]
2022-01-20 18:58:27,768 > Successfully built OctoPrint-WemoSwitch
2022-01-20 18:58:27,769 > Installing collected packages: OctoPrint-WemoSwitch
2022-01-20 18:58:27,770 > Attempting uninstall: OctoPrint-WemoSwitch
2022-01-20 18:58:27,770 > Found existing installation: OctoPrint-WemoSwitch 0.1.10
2022-01-20 18:58:27,770 > Uninstalling OctoPrint-WemoSwitch-0.1.10:
2022-01-20 18:58:27,770 > Successfully uninstalled OctoPrint-WemoSwitch-0.1.10
2022-01-20 18:58:27,771 > Successfully installed OctoPrint-WemoSwitch-0.1.10`

Thanks for your help.

jneilliii, 2022-01-20:
Check octoprint.log for the two errors mentioned in the pinned issues...

mdubash, 2022-01-20:
I'm picking up this error in octoprint.log:
--
'2022-01-20 16:55:17,834 - octoprint.plugin.core - ERROR - Error loading plugin wemoswitch
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_wemoswitch/__init__.py", line 16, in <module>
2022-01-20 18:58:17,214 - octoprint.plugin.core - ERROR - Error loading plugin wemoswitch
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_wemoswitch/__init__.py", line 16, in <module>
2022-01-20 18:58:28,379 - octoprint.plugin.core - ERROR - Error loading plugin wemoswitch
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_wemoswitch/__init__.py", line 16, in <module>
2022-01-20 18:58:42,458 - octoprint.plugin.core - ERROR - Error loading plugin wemoswitch
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_wemoswitch/__init__.py", line 16, in <module>
2022-01-20 22:56:45,439 - octoprint.plugin.core - ERROR - Error loading plugin wemoswitch
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_wemoswitch/__init__.py", line 16, in <module>
2022-01-20 22:56:56,700 - octoprint.plugin.core - ERROR - Error loading plugin wemoswitch
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_wemoswitch/__init__.py", line 16, in <module>
2022-01-20 22:58:05,283 - octoprint.plugin.core - ERROR - Error loading plugin wemoswitch
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_wemoswitch/__init__.py", line 16, in <module>'
--

The full log is here:
'2022-01-20 13:59:57,310 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2022-01-20 13:59:57,318 - octoprint.server - INFO - --- Log roll over detected ---------------------------------------------------
2022-01-20 13:59:57,319 - octoprint.server - INFO - OctoPrint 1.7.2
2022-01-20 13:59:57,326 - octoprint.plugin.core - INFO - 38 plugin(s) registered with the system:
|  Access Anywhere - The Spaghetti Detective (1.8.9) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_thespaghettidetective
|  Action Command Notification Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_notification
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/appkeys
|  Arc-Welder (1.0.0+u.bb71e8f) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_arc_welder
|  Autologin Configuration (0.2.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_autologin_config
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/backup
|  Bed Visualizer (1.1.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_bedlevelvisualizer
|  BetterHeaterTimeout (1.3.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_BetterHeaterTimeout
|  Core Wizard (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/corewizard
|  Dashboard (1.19.6) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_dashboard
|  DeleteAfterPrint Plugin (1.10.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_DeleteAfterPrint
|  Discovery (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/discovery
|  DisplayLayerProgress Plugin (1.27.2) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_DisplayLayerProgress
|  Error Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/errortracking
|  Event Manager (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/eventmanager
|  Exclude Region (0.3.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_excluderegion
| !Filament Manager (1.9.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_filamentmanager
|  File Check (2021.2.23) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_file_check
|  Firmware Check (2021.10.11) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_firmware_check
|  Firmware Updater (1.13.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_firmwareupdater
|  Fullscreen Plugin (0.0.6) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_fullscreen
|  GCode Viewer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/gcodeviewer
|  Logging (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/logging
| *OctoPrint-AutomaticShutdown (0.1.4) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_automaticshutdown
|  OctoPrint-IFTTT (1.4.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_IFTTT
|  Pi Support Plugin (2021.10.28) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/pluginmanager
|  Preheat Button (0.8.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_preheat
|  PrintTimeGenius Plugin (2.2.8) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius
|  Procastinator (1.1.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_procastinator
|  Simple Emergency Stop (1.0.5) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_simpleemergencystop
[... cut ...] egion': [{"command": "ExcludeRegion", "parameterPattern": "^\\s*(enable|on)(\\s|$)", "action": "enable_exclusion", "description": "Default action to enable exclusion", "type": "AtCommandAction"}, {"command": "ExcludeRegion", "parameterPattern": "^\\s*(disable|off)(\\s|$)", "action": "disable_exclusion", "description": "Default action to disable exclusion", "type": "AtCommandAction"}]}
2022-01-20 22:49:25,205 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.254.36'

jneilliii, 2022-01-21:
> ImportError: libxslt.so.1: cannot open shared object file: No such file or directory

There is a pinned issue about this specific error. Check the solution provided at the end of this post.

https://github.com/jneilliii/OctoPrint-WemoSwitch/issues/51#issuecomment-846527793

closed 2022-01-21 by jneilliii