jneilliii/OctoPrint-WemoSwitch #61: Provide alias/label for discovered device to handle port changes

swimmadude66, 2022-03-10:
My Wemo plug seems to like to change ports pretty regularly. A cursory google search shows this is a common issue, but can only really be fixed at the integration layer by trying multiple ports. 

Since the `Discovered Devices` dropdown in the settings for this plugin always seems to find the device, is there any way we could configure the plugin to select the device from the discovered list (with whatever port it is currently using) by MAC address or hostname?

The port sometimes changes mid-print, meaning the plugin can turn the printer on, but not off, or vice-versa

jneilliii, 2022-03-10:
I hope you mean IP address and not port? If that is the case then you can use hostname of the device and the plugin will try to resolve the IP address from that [here](https://github.com/jneilliii/OctoPrint-WemoSwitch/blob/b59b4eb0b5ecc7f1cbf1e5defcc81bf70d511781/octoprint_wemoswitch/__init__.py#L567). Another option might be to make a DHCP reservation on your router for that device so it always gets the same IP address. This is what I typically do with my IoT devices I want to control outside of their own ecosystem.

swimmadude66, 2022-03-11:
I do not, I mean port. The device has a static IP on my network, but the port jumps between 49154 and 49153, seemingly randomly.

OpenHab and others have had to address this behavior: https://community.openhab.org/t/solved-wemo-bounty-to-resolve-port-changing/74453

jneilliii, 2022-03-11:
Oh, that's weird. I've never used the wemo long enough to come across that or the one I have doesn't have the issue. But yeah, I can attempt to work in something like they did [here](https://github.com/openhab/openhab-addons/pull/6253/files#diff-c15fe4ffad484dc16712667747162145527f62bc97a9964ed32c503bb0ff6e96R464-R482) and just try the pot ranges until successful. Thanks for the pointer.

swimmadude66, 2022-03-11:
thank you! The reason for wanting an alias or hostname reference is that the GCODE command still currently requires `ip:port` which is difficult when the port changes so much

jneilliii, 2022-03-11:
I was just reviewing the code and if there is no port specified it tries to poll the device for it's info by IP address automatically, so without the :port the gcode command doesn't need it I don't think. It's been a while since I worked on this plugin though, so I might be wrong.

swimmadude66, 2022-03-16:
awesome, thank you! I will try that out

swimmadude66, 2022-03-16:
that seems to work exactly as needed. thanks!

closed 2022-03-16 by swimmadude66