jneilliii/OctoPrint-WemoSwitch #64: Wemo switch often reported as on while it's off, requires toggling off/on in UI to get on again

epheterson, 2022-08-01:
Sometimes I go to print and I see the bolt is green while the switch is actually off. No amount of waiting updates it to the correct state, I have to toggle it off in the UI, then back on, and it always turns on fine.

Any idea what could be causing this issue or how to fix it? I've tried decreasing the polling interval but saw no change.

jneilliii, 2022-08-02:
hard to say. if you enable debug logging in the plugin settings and restart OctoPrint it might help us see what's happening, or at least what the plugin is doing. Polling status should definitely be handling this state change from outside the plugin.

epheterson, 2022-08-02:
I just noticed it's in this state. I went to print something earlier today and it correctly showed as red/off while it was off. I then tried rebooting Octoprint, and still it correctly showed as red/off. 

After the print completed, it shows in the logs that it turned the switch off, and it is indeed off, though the bolt is currently green/on. In this state, reloading the web UI and relaunching OctoPrint shows "Waiting..." then puts the green bolt there, so something isn't reading properly.

See debug logs –
[plugin_wemoswitch_debug-2.log](https://github.com/jneilliii/OctoPrint-WemoSwitch/files/9246319/plugin_wemoswitch_debug-2.log)

epheterson, 2022-08-17:
Can you find anything useful in the logs? Anything else I can try? It’d be nice if the “Turn Off” pop-up had a Turn On button (maybe even optionally) since when we eventually send an On command, that does work.

jneilliii, 2022-08-17:
Sorry, I've been busy with other stuff and haven't had a chance to look yet.

jneilliii, 2022-08-18:
So I got a chance to look at the log and there doesn't seem to be anything out of the ordinary. What it sounds like is the state in the UI is getting out of sync with the server side knowledge of the device's current state. Do you happen to show any errors in the browser's developer tools (press F12 on your keyboard) on the console tab on page load when it is like this?

jneilliii, 2022-08-19:
the 404 ones I've seen before for fontawesome webfont download, but that shouldn't impact the plugin. the websocket one is strange, can't say I've seen that before and might be part of a chrome extension possibly.

epheterson, 2022-12-08:
I enabled Web Sockets on my reverse proxy and that improved the behavior of Octoprint overall (as well as many other self hosted apps) and seems to have made this issue disappear too. So, closing and thanks!

closed 2022-12-08 by epheterson