jneilliii/OctoPrint-WemoSwitch #65: Controlling Wemo in Safari 15.6.1 dosnt work

nbaddorf, 2022-10-01:
I have been messing around with this plugin trying to get it to work without success. I finally tryed controling the wemo on my iphone through octoprint and it works great. I tried Chrome and it works, although if the wemo is on, the status showes off untill you toggle it. 
The button dosnt work in Safari. The lightning bolt symbol spins but never works.

Thanks

jneilliii, 2022-10-01:
Could you check safari's developer tools to see if there are any errors that might shed some light into why it doesn't work there? I don't have access to iOS/Apple devices to develop against.

nbaddorf, 2022-10-02:
Hi, thanks for the quick response! When I go to inspect the webpage, the consol gives me this error: WebSocket connection to 'ws://octopi.local/sockjs/802/ion2ocj2/websocket' failed: The operation couldn’t be completed. Socket is not connected
Selected Element
<div class="page-container" id="page-container-main" style data-bind="visible: visible">…</div>

Clicking the lightning button only seems to work when the wemo is currently on and the lightning symbol is on, which turns the wemo off. However, the lightning symbol always spins.

jneilliii, 2022-10-02:
Ok, that could be related error, but would be an underlying problem with OctoPrint and establishing the communication layer between the front end and server-side backend. It's that communication layer that the plugin uses to tell the front end the results of the on/off commands. If that message never makes it then the plugin doesn't know what to do. Are there any interesting errors in the logs around the time you were attempting the power control?

nbaddorf, 2022-10-02:
Here is a sample of the logs I got for the wemo:
`[2022-10-01 15:22:29,859] DEBUG: IP 192.168.86.47 is valid.
[2022-10-01 15:22:29,860] DEBUG: Attempting to connect to 192.168.86.47
[2022-10-01 15:22:29,885] DEBUG: Getting device info from http://192.168.86.47:49153/setup.xml
[2022-10-01 15:22:30,201] DEBUG: Found device <WeMo Switch "Ender 3">
[2022-10-01 15:22:30,202] DEBUG: Sending command on to 192.168.86.47
[2022-10-01 15:22:30,248] DEBUG: Checking status of 192.168.86.47.
[2022-10-01 15:22:30,248] DEBUG: IP 192.168.86.47 is valid.
[2022-10-01 15:22:30,248] DEBUG: Attempting to connect to 192.168.86.47
[2022-10-01 15:22:30,278] DEBUG: Getting device info from http://192.168.86.47:49153/setup.xml
[2022-10-01 15:22:30,589] DEBUG: Found device <WeMo Switch "Ender 3">
[2022-10-01 15:22:30,590] DEBUG: Sending command info to 192.168.86.47
[2022-10-01 15:22:36,251] DEBUG: Discovering devices
`

It was talking to the wemo but never controlling it. Again, every other device I have tried worked.