jneilliii/OctoPrint-WemoSwitch #68: Seems to be working to control WeMo plugs, but icons are not changing color

jonheese-wti, 2023-04-14:
I searched through all of the old issues but I'm not seeing anyone else report this issue.

I just installed the plugin today to use with my WeMo switches, and it seems to be working correctly -- I can turn the switches on and off without a problem -- but the icon in my Octoprint UI is always gray.

I looked at the code and I don't see where the `btnColor` property is set to anything but `#808080` so I'm not sure off-hand where the problem might lie.

Any tips for how to troubleshoot this?  Thanks.

jneilliii, 2023-04-14:
If you open your developer tools in the browser and press the button do you get any errors on the console tab?

jonheese-wti, 2023-04-14:
No errors when I click the buttons, but I get this every time the page loads:

```
packed_libs.js?7111420c:13     POST https://octoprint.jonheese.local/api/plugin/wemoswitch 500 (INTERNAL SERVER ERROR)
send @ packed_libs.js?7111420c:13
ajax @ packed_libs.js?7111420c:13
self.checkStatus @ packed_plugins.js?081422ed:12545
(anonymous) @ packed_plugins.js?081422ed:12564
D @ packed_libs.js?7111420c:129
self.checkStatuses @ packed_plugins.js?081422ed:12561
self.onAfterBinding @ packed_plugins.js?081422ed:12374
callViewModelIf @ packed_core.js?389582e5:1268
callViewModel @ packed_core.js?389582e5:1264
(anonymous) @ packed_core.js?389582e5:1338
Pn @ packed_libs.js?7111420c:30
(anonymous) @ packed_libs.js?7111420c:51
bindViewModels @ packed_core.js?389582e5:1322
setTimeout (async)
(anonymous) @ packed_core.js?389582e5:1341
c @ packed_libs.js?7111420c:13
fireWith @ packed_libs.js?7111420c:13
l @ packed_libs.js?7111420c:13
(anonymous) @ packed_libs.js?7111420c:13
load (async)
send @ packed_libs.js?7111420c:13
ajax @ packed_libs.js?7111420c:13
OctoPrintClient.ajax @ packed_client.js?e6252517:12
OctoPrintClient.get @ packed_client.js?e6252517:12
OctoPrintSettingsClient.get @ packed_client.js?e6252517:110
SettingsViewModel.self.requestData @ packed_core.js?389582e5:336
fetchSettings @ packed_core.js?389582e5:1341
setTimeout (async)
(anonymous) @ packed_core.js?389582e5:1341
c @ packed_libs.js?7111420c:13
fireWith @ packed_libs.js?7111420c:13
l @ packed_libs.js?7111420c:13
(anonymous) @ packed_libs.js?7111420c:13
load (async)
send @ packed_libs.js?7111420c:13
ajax @ packed_libs.js?7111420c:13
OctoPrintClient.ajax @ packed_client.js?e6252517:12
OctoPrintClient.ajaxWithData @ packed_client.js?e6252517:12
OctoPrintClient.post @ packed_client.js?e6252517:12
OctoPrintClient.postJson @ packed_client.js?e6252517:12
OctoPrintBrowserClient.passiveLogin @ packed_client.js?e6252517:59
LoginStateViewModel.self.requestData @ packed_core.js?389582e5:229
onServerConnect @ packed_core.js?389582e5:1341
(anonymous) @ packed_core.js?389582e5:1341
c @ packed_libs.js?7111420c:13
fireWith @ packed_libs.js?7111420c:13
s.<computed> @ packed_libs.js?7111420c:13
DataUpdater.self._onConnectMessage @ packed_core.js?389582e5:1136
(anonymous) @ packed_client.js?e6252517:24
Pn @ packed_libs.js?7111420c:30
(anonymous) @ packed_libs.js?7111420c:51
OctoPrintSocketClient.propagateMessage @ packed_client.js?e6252517:24
(anonymous) @ packed_client.js?e6252517:31
(anonymous) @ packed_libs.js?7111420c:47
_t @ packed_libs.js?7111420c:35
(anonymous) @ packed_libs.js?7111420c:46
(anonymous) @ packed_libs.js?7111420c:51
onMessage @ packed_client.js?e6252517:31
r.dispatchEvent @ packed_libs.js?7111420c:1585
(anonymous) @ packed_libs.js?7111420c:1585
w._transportMessage @ packed_libs.js?7111420c:1585
i.emit @ packed_libs.js?7111420c:1585
ws.onmessage @ packed_libs.js?7111420c:1585
```

jneilliii, 2023-04-14:
Thanks, download octoprint.log from the logging section of settings and share that here please (drag the file into comment area).

jonheese-wti, 2023-04-14:
[octoprint.log](https://github.com/jneilliii/OctoPrint-WemoSwitch/files/11236433/octoprint.log)

jneilliii, 2023-04-14:
> 2023-04-14 13:35:51,000 - pywemo.discovery - ERROR - Failed to fetch description http://192.168.27.204:49153/setup.xml

Curious, if your wemo device has changed IP addresses possibly and isn't 192.168.27.204 anymore?

jneilliii, 2023-04-14:
Thanks, I just installed on a fresh OctoPrint instance and am seeing the same thing as you with respect to it working but not changing the status of the icon. I suspect something may have changed in the pywemo library that has broken this plugin. I'll see if I can debug and determine what that is over the weekend.

jneilliii, 2023-04-14:
Ah, I think I found the issue. I had the wrong filename reference for the css file. Could you try installing the patched version and verify on your end as well before I release? Go into Plugin Manager and click the get more button and at the bottom of the window that opens in the ...from URL field copy/paste the following and click install.

```
https://github.com/jneilliii/OctoPrint-WemoSwitch/archive/refs/heads/bugfix/css.zip
```

Once verified I can merge the changes into stable release and push out a new version.

jonheese-wti, 2023-04-14:
Yes, that fixed it.  Thanks!

closed 2023-04-14 by jneilliii