jneilliii/OctoPrint-WemoSwitch #70: Plug-in does not discover devices on the network

DaHouzKat, 2023-07-01:
Hi,

I have difficulty setting up the plug-in for my new printer. 

On my old OctoPrint installation i have the drop-down list that allows me to choose between the Wemo plug connected to my network and everything is working properlly.

But on my new OctoPrint installation i don't see any drop down, all i see is "Trying to discover devices on the network".  

I tried to manually add the IP Address i see in my old installation but it does not work.

I checked in the log and it looks like it sees it but saddly it does not work :-(

[2023-07-01 10:31:15,319] DEBUG: Discovering devices
[2023-07-01 10:31:21,630] DEBUG: Checking status of 192.168.4.6:49152.
[2023-07-01 10:31:21,631] DEBUG: IP 192.168.4.6 is valid.
[2023-07-01 10:31:21,631] DEBUG: Attempting to connect to 192.168.4.6
[2023-07-01 10:31:21,631] DEBUG: Getting device info from http://192.168.4.6:49152/setup.xml

My New Installation : Octoprint 1.9.1 / OctoPi 1.0.0 / RPi 4 Model B Rev 1.5

[image]

My Old Installation :  Octoprint 1.9.1 / OctoPi 0.18.0 / RPi 4 Model B Rev 1.2

[image]

jneilliii, 2023-07-01:
The discovery and communication to wemo is being handled by a python module I didn't create. If it's not working then either it's broken or wemo changed something that broker it. Do you have any other entries in octoprint.log that might indicate a potential error in the module?

DaHouzKat, 2023-07-01:
All my wemo switches were bought together two years ago so i don't think it as something to do with the plugs...  But since the OctoPi version are not the same...  maybe there's something wrong with the new installation...  i will look into the Octoprint log as you suggested...  Thx !!

DaHouzKat, 2023-07-01:
You are right, i thing there's is something wrong with my OctoPi/OctoPrint-Python Installation...

2023-07-01 11:42:56,346 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin wemoswitch
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/server/api/__init__.py", line 159, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/util/__init__.py", line 1686, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint_wemoswitch/__init__.py", line 324, in on_api_command
    self.check_status("{ip}".format(**data))
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint_wemoswitch/__init__.py", line 289, in check_status
    chk = self.sendCommand("info", plugip)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint_wemoswitch/__init__.py", line 597, in sendCommand
    device = pywemo.discovery.device_from_description(url, None)
TypeError: device_from_description() takes 1 positional argument but 2 were given

DaHouzKat, 2023-07-01:
My old RPI is using python 3.7.3 instead of python 3.9.2
[image]

Did you think i should i revert to python 3.7.3 ?

jneilliii, 2023-07-01:
I doubt that is necessary. I just need to check to see what changed in the upstream pywemo module and adjust accordingly.

jneilliii, 2023-07-01:
can you do me a favor and run the following command on both instances you have. Might help me isolate the version that this changed. 

`~/oprint/bin/pip freeze | grep wemo`

DaHouzKat, 2023-07-01:
of course...  Thx for your help !!!
[image]

jneilliii, 2023-07-01:
perfect, so what's happened is the upstream module changed. if you run `~/oprint/bin/pip install --force-reinstall pywemo==0.7.0` on the broken instance it will probably fix it. otherwise, I just made the change for 1.0.0 that should also correct it. In plugin manager go to get more, and at the bottom in the ...from URL field copy/paste the below and click install and see if that fixes it. 

```
https://github.com/jneilliii/OctoPrint-WemoSwitch/archive/refs/heads/bugfix/pywemo_discovery.zip
```

DaHouzKat, 2023-07-01:
wow, that was fast !!!

is there one of the 2 solutions you would like me to try first ?

DaHouzKat, 2023-07-01:
wow that was fast !!!

The "pip install..." fixed my problem !!!

Wemo Plug-in now works like a charm on both of my printers !!!

Thank you very much for your help !!!

closed 2023-07-01 by DaHouzKat

jneilliii, 2023-07-01:
thanks for confirming. It looks like it was recently released version 1.0.0 of pywemo that removed the argument and started causing the error. I have pinned the version dependency to less than 1.0.0 to hopefully prevent this from happening to others in the future.