jneilliii/OctoPrint-WemoSwitch #71: idle timeout doesn't work

DaHouzKat, 2023-07-04:
Hi,

I managed to get the OctoPrint / Wemo Switch to work with my Prusa Mk4, but the 20 minute idle timeout doesn't work... 

The setup is pretty much like my Prusa Mini which works perfectly. I compared the logs but didn't see anything unusual. 

Any idea where i should look ?!

jneilliii, 2023-07-04:
I bet it's a similar issue I recently fixed in TP-Link plugin for the mk4. I suspect you are using the idle timeout to power off? Prusa changed their temperature reporting to include an X value that is detected and can never turn off.

DaHouzKat, 2023-07-04:
Yes i'm using the idle timeout to power off.

I did not think of comparing the content of the messages, you are right Prusa change their temperature reporting :-(

> Mk4 -  Recv:  T:169.81/170.00 B:60.55/60.00 X:32.88/36.00 A:40.67/0.00 @:43 B@:32 HBR@:55
> Mini -  Recv:  T:210.05 /210.00 B:59.99 /60.00 A:-30.00 /0.00 @:67 B@:40

Is there a way to fix that or a work around i could use ?

jneilliii, 2023-07-04:
go into software update in OctoPrint's settings and do a force update check and upgrade to version 0.1.13 where I've added an Ignored Heaters option in the idle timeout settings. In that field enter the letter `X` like below.

[image]

closed 2023-07-04 by jneilliii

DaHouzKat, 2023-07-04:
It works now !!!  Thank you very much for your help !!!