jneilliii/OctoPrint-WemoSwitch #72: Pywemo >=1.0.0" error during update

bakerlogan, 2023-07-04:
Hi, 
I attempted to update this plugin and get the following error related to a pywemo version >= 1.0.0 not being found. 

Based on the error it seems octoprint doesn't recognize the newest versions of pywemo yet

> ++++++++++++++++++++++++++++++++++
> Now updating Wemo Switch to 0.1.13
> ++++++++++++++++++++++++++++++++++
> /home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install https://github.com/jneilliii/OctoPrint-WemoSwitch/archive/0.1.13.zip --no-cache-dir
> Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
> Collecting https://github.com/jneilliii/OctoPrint-WemoSwitch/archive/0.1.13.zip
> Downloading https://github.com/jneilliii/OctoPrint-WemoSwitch/archive/0.1.13.zip
> Preparing metadata (setup.py): started
> Preparing metadata (setup.py): finished with status 'done'
> Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-WemoSwitch==0.1.13) (1.9.1)
> ERROR: Could not find a version that satisfies the requirement pywemo>=1.0.0 (from octoprint-wemoswitch) (from versions: 0.4.46.macosx-10.15-x86_64, 0.1, 0.1.1, 0.2, 0.3, 0.3.1, 0.3.2, 0.3.3, 0.3.4, 0.3.5, 0.3.6, 0.3.7, 0.3.8, 0.3.9, 0.3.10, 0.3.11, 0.3.12, 0.3.13, 0.3.14, 0.3.15, 0.3.16, 0.3.17b0, 0.3.17, 0.3.18, 0.3.19, 0.3.20, 0.3.21, 0.3.22, 0.3.23, 0.3.24, 0.4.1, 0.4.2, 0.4.3, 0.4.4, 0.4.5, 0.4.6, 0.4.7, 0.4.8, 0.4.9, 0.4.10, 0.4.11, 0.4.12, 0.4.13, 0.4.14, 0.4.15, 0.4.16, 0.4.17, 0.4.18, 0.4.19, 0.4.20, 0.4.21, 0.4.22, 0.4.23, 0.4.24, 0.4.25, 0.4.26, 0.4.27, 0.4.28, 0.4.29, 0.4.30, 0.4.32, 0.4.33, 0.4.34, 0.4.35, 0.4.36, 0.4.37, 0.4.38, 0.4.39, 0.4.40, 0.4.41, 0.4.42, 0.4.43, 0.4.44, 0.4.45, 0.4.46, 0.5.0, 0.5.2, 0.5.3, 0.5.4, 0.5.6, 0.6.1, 0.6.2, 0.6.3, 0.6.4, 0.6.5, 0.6.6, 0.6.7, 0.6.8, 0.7.0b1, 0.7.0, 0.8.0, 0.8.1, 0.9.0, 0.9.1, 0.9.2)
> ERROR: No matching distribution found for pywemo>=1.0.0
> 
> The update did not finish successfully. Please consult <code>octoprint.log</code> and <code>plugin_softwareupdate_console.log</code> for details.

jneilliii, 2023-07-05:
What version octopi/python versions are you running?

jneilliii, 2023-07-05:
just changed the version pinning in 0.1.14, which will support >=0.8.0 versions of pywemo. Seems python 3.7 support was dropped in version 1.0.0.

closed 2023-07-05 by jneilliii