jneilliii/OctoPrint-WemoSwitch #76: Fresh OctoPrint installation - Can't Install Plugin

AngryKiwiNZ, 2025-04-28:
After a failure on my older Octoprint installation I've installed the latest 1.11.0 and restored my backup. I'm now getting the following error when trying to add the WeMo plugin:

I've run an apt-get update and upgrade and confirmed everything is up to date, but the issue persists.

Installing plugin "OctoPrint-WemoSwitch" from https://github.com/jneilliii/OctoPrint-WemoSwitch/archive/master.zip...
/home/jay/oprint/bin/python -m pip --disable-pip-version-check install file:///tmp/tmpz4c3d7bc/OctoPrint-WemoSwitch-master.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmpz4c3d7bc/OctoPrint-WemoSwitch-master.zip
Preparing metadata (setup.py): started
Preparing metadata (setup.py): finished with status 'done'
Requirement already satisfied: OctoPrint in /home/jay/oprint/lib/python3.9/site-packages (from OctoPrint-WemoSwitch==0.1.14) (1.11.0)
Collecting pywemo>=0.8.0
Downloading https://www.piwheels.org/simple/pywemo/pywemo-1.4.0-py3-none-any.whl (101 kB)
━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 101.1/101.1 kB 151.3 kB/s eta 0:00:00
Collecting uptime
Downloading https://www.piwheels.org/simple/uptime/uptime-3.0.1-cp39-cp39-linux_armv7l.whl (14 kB)
Requirement already satisfied: requests>=2.0 in /home/jay/oprint/lib/python3.9/site-packages (from pywemo>=0.8.0->OctoPrint-WemoSwitch==0.1.14) (2.32.3)
Collecting lxml>=4.6
Downloading lxml-5.4.0.tar.gz (3.7 MB)
━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.7/3.7 MB 5.5 MB/s eta 0:00:00
Installing build dependencies: started
Installing build dependencies: finished with status 'done'
Getting requirements to build wheel: started
Getting requirements to build wheel: finished with status 'error'
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> [4 lines of output]
<string>:67: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
Building lxml version 5.4.0.
Building without Cython.
Error: Please make sure the libxml2 and libxslt development packages are installed.
[end of output]

note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
Looks like the plugin was already installed. Forcing a reinstall.
/home/jay/oprint/bin/python -m pip --disable-pip-version-check install file:///tmp/tmpz4c3d7bc/OctoPrint-WemoSwitch-master.zip --no-cache-dir --ignore-installed --force-reinstall --no-deps
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmpz4c3d7bc/OctoPrint-WemoSwitch-master.zip
Preparing metadata (setup.py): started
Preparing metadata (setup.py): finished with status 'done'
Building wheels for collected packages: OctoPrint-WemoSwitch
Building wheel for OctoPrint-WemoSwitch (setup.py): started
Building wheel for OctoPrint-WemoSwitch (setup.py): finished with status 'done'
Created wheel for OctoPrint-WemoSwitch: filename=OctoPrint_WemoSwitch-0.1.14-py3-none-any.whl size=298667 sha256=c88dd0f02e08a77c0707c9a97dd0f009861cc3803156a15e72a534245f7aa846
Stored in directory: /tmp/pip-ephem-wheel-cache-fsfl4atc/wheels/ce/ef/08/caa3bd58043f65f2f4ac4b31fa6a1d8659ab1d0fa14bcec995
Successfully built OctoPrint-WemoSwitch
Installing collected packages: OctoPrint-WemoSwitch
Successfully installed OctoPrint-WemoSwitch-0.1.14
Done!

jneilliii, 2025-04-28:
> Please make sure the libxml2 and libxslt development packages are installed.

Potential answer in pinned post...

https://github.com/jneilliii/OctoPrint-WemoSwitch/issues/51#issuecomment-846527793

AngryKiwiNZ, 2025-04-28:
> > Please make sure the libxml2 and libxslt development packages are installed.
> 
> Potential answer in pinned post...
> 
> [#51 (comment)](https://github.com/jneilliii/OctoPrint-WemoSwitch/issues/51#issuecomment-846527793)

Thank you, I've manually installed the files which has allowed the installation of the WeMo plugin to begin which looks promising, but sadly it never ends, and it looks like I don't have sufficient RAM on my Pi Zero 2W for the process to complete. I left it for an hour without any joy!

Both physical memory and the swap file are exhausted and even staying connected via SSH is a challenge....

Installing plugin "OctoPrint-WemoSwitch" from https://github.com/jneilliii/OctoPrint-WemoSwitch/archive/master.zip...
/home/jay/oprint/bin/python -m pip --disable-pip-version-check install file:///tmp/tmpvliw40s3/OctoPrint-WemoSwitch-master.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmpvliw40s3/OctoPrint-WemoSwitch-master.zip
Preparing metadata (setup.py): started
Preparing metadata (setup.py): finished with status 'done'
Requirement already satisfied: OctoPrint in /home/jay/oprint/lib/python3.9/site-packages (from OctoPrint-WemoSwitch==0.1.14) (1.11.0)
Collecting pywemo>=0.8.0
Downloading https://www.piwheels.org/simple/pywemo/pywemo-1.4.0-py3-none-any.whl (101 kB)
━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 101.1/101.1 kB 145.4 kB/s eta 0:00:00
Collecting uptime
Downloading https://www.piwheels.org/simple/uptime/uptime-3.0.1-cp39-cp39-linux_armv7l.whl (14 kB)
Requirement already satisfied: urllib3>=1.26.0 in /home/jay/oprint/lib/python3.9/site-packages (from pywemo>=0.8.0->OctoPrint-WemoSwitch==0.1.14) (1.26.12)
Collecting lxml>=4.6
Downloading lxml-5.4.0.tar.gz (3.7 MB)
━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.7/3.7 MB 3.5 MB/s eta 0:00:00
Installing build dependencies: started
Installing build dependencies: finished with status 'done'
Getting requirements to build wheel: started
Getting requirements to build wheel: finished with status 'done'
Preparing metadata (pyproject.toml): started
Preparing metadata (pyproject.toml): finished with status 'done'
Requirement already satisfied: ifaddr>=0.1.0 in /home/jay/oprint/lib/python3.9/site-packages (from pywemo>=0.8.0->OctoPrint-WemoSwitch==0.1.14) (0.2.0)
Requirement already satisfied: requests>=2.0 in /home/jay/oprint/lib/python3.9/site-packages (from pywemo>=0.8.0->OctoPrint-WemoSwitch==0.1.14) (2.32.3)
Requirement already satisfied: wrapt<1.18,>=1.17.2 in /home/jay/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.14) (1.17.2)
Requirement already satisfied: Flask-Limiter<3.9,>=3.8 in /home/jay/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.14) (3.8.0)
Requirement already satisfied: wheel in /home/jay/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.14) (0.34.2)
Requirement already satisfied: libpass<2,>=1.8.1 in /home/jay/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.14) (1.9.0)
Requirement already satisfied: psutil<7,>=6.1.1 in /home/jay/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.14) (6.1.1)
Requirement already satisfied: Flask-Login<0.7,>=0.6.3 in /home/jay/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.14) (0.6.3)
Requirement already satisfied: pytz in /home/jay/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.14) (2025.2)
Requirement already satisfied: Babel<2.17,>=2.16 in /home/jay/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.14) (2.16.0)
Requirement already satisfied: sarge==0.1.7.post1 in /home/jay/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.14) (0.1.7.post1)
Requirement already satisfied: more-itertools>=10.5.0 in /home/jay/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.14) (10.6.0)
Requirement already satisfied: colorlog<7,>=6.9.0 in /home/jay/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.14) (6.9.0)
Requirement already satisfied: emoji<3,>=2.14.1 in /home/jay/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.14) (2.14.1)
Requirement already satisfied: pydantic<3,>=2.10.6 in /home/jay/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.14) (2.11.3)
Requirement already satisfied: sentry-sdk<3,>=2.20.0 in /home/jay/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.14) (2.26.1)
Requirement already satisfied: werkzeug<3.1,>=3.0.6 in /home/jay/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.14) (3.0.6)
Requirement already satisfied: pathvalidate<4,>=3.2.1 in /home/jay/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.14) (3.2.3)
Requirement already satisfied: filetype<2,>=1.2.0 in /home/jay/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.14) (1.2.0)
Requirement already satisfied: setuptools in /home/jay/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.14) (44.1.1)
Requirement already satisfied: netaddr<1.4,>=1.3.0 in /home/jay/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-WemoSwitch==0.1.14) (1.3.0)
[... cut ...]
Building wheels for collected packages: OctoPrint-WemoSwitch, lxml
Building wheel for OctoPrint-WemoSwitch (setup.py): started
Building wheel for OctoPrint-WemoSwitch (setup.py): finished with status 'done'
Created wheel for OctoPrint-WemoSwitch: filename=OctoPrint_WemoSwitch-0.1.14-py3-none-any.whl size=298667 sha256=58d028035fe3d638e3dfe86057aca8ee0b1041bba20485d0c5aa44c49bc22a7b
Stored in directory: /tmp/pip-ephem-wheel-cache-h6egzpk5/wheels/89/1c/66/43c809636582b64fd2a0a8ac76f477277ba4e2a3bc019e64a1
Building wheel for lxml (pyproject.toml): started

jneilliii, 2025-04-28:
Ah, ok. In that case you may have to use swap memory to get the install to complete. I've had to do that before with lower memory pi devices for one of my other plugins. See if this makes a difference in your situation.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/141#issuecomment-760266240

AngryKiwiNZ, 2025-04-28:
> Ah, ok. In that case you may have to use swap memory to get the install to complete. I've had to do that before with lower memory pi devices for one of my other plugins. See if this makes a difference in your situation.
> 
> [jneilliii/OctoPrint-BedLevelVisualizer#141 (comment)](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/141#issuecomment-760266240)

Thank you. I will look up how to extend the size of the page file to see if I can allow the process to complete.

Thank you kindly for replies and help, I really appreciate it!

AngryKiwiNZ, 2025-04-28:
> > I will look up how to extend the size of the page
> 
> that comment link I posted has the commands to run on your pi to do it.

Thanks again, doing this now. Has so far been running for 52 minutes with 1GB Page File and it's still grinding it's way through!!

AngryKiwiNZ, 2025-04-28:
I'm so happy it works again now! - It finished installing on the R-Pi Zero 2W after 1 hour and 10 minutes!

Thanks again for your support. Much appreciated!!

jneilliii, 2025-04-29:
Great news, glad we were able to get through it.

closed 2025-04-29 by jneilliii